teckel12/LuaTelemetry #346: Saying whatever

mistyk, 2019-09-11:
I'm using this script on my Taranis QX7 with inav 2.1, sometime taranis speak "warning battery 10545% or failsafe enable or disarmed at any time without reasons...

do you know where it comes from? all my friends have the same issue...

teckel12, 2019-09-11:
@mistyk It's just reporting the telemetry that's being received.  My guess is that you're running Crossfire. I'm guessing that because Crossfire sends a lot of bad telemetry data. If you were using FrSky S.Port telemetry, you'd never have an issue.

Anyway, if you're using Crossfire, the issue is Crossfire telemetry. It will sometimes just not send a flight mode, or other empty or junk data. Lua Telemetry can only assume its real, and has no way of knowing otherwise.

The fuel message can probably be solved by turning on the `Filter` option on the `Capa` sensor.

Also, running the latest development branch (v1.7.3) may fix things too, as a change was made to try and dectect cases were Crossfire telemetry is partially lost, but hasn't been detected yet.

Try the dev branch, set the filter option, and let me know if that makes it better. In any case, the core of the issue is Crossfire reporting bad telemetry data that Lua Telemetry is just reporting to you.

mistyk, 2019-09-11:
Thanks for your speed reply, i'm not using crossfire but R9MM from Frsky S.Port (pretty near)  (Five people have the same R9MM with the same issue).
I'll look your Filter  option, we'll see ! Thx !

teckel12, 2019-09-11:
@mistyk I guess it could be an issue with the R9MM. Again, Lua Telemetry is just reporting what is being received.  I used alot of FrSky receivers and I've never had the issues you're reporting.

I'd like to see the telemetry logs where this happens. Also, your transmitter settings etc as detailed in the issue template. With this, I can playback your flight and see what's happening.

Are you doing long range?

mistyk, 2019-09-12:
Yes i'm doing long range, i've well understood the problem, i'll send to you my dat file at next flight and transmitter settings.
Thanks !

teckel12, 2019-09-23:
@mistyk I would like to see telemetry logs of this happening.  I could then run the simulator and playback your flight to diagnose the problem.

teckel12, 2019-10-01:
@mistyk What's the telemetry update speed with R9?  Another user is reporting VERY slow telemetry updates (like once every 10 seconds).  See: #351

closed 2019-11-22 by stale[bot]