teckel12/LuaTelemetry #348: Add another option for Voice Alerts

Jetrell, 2019-09-17:
Hi Tim,
At present the Voice Alerts only have -  `ALL` and `Critical`.

The reason I see this as an issue. Is because the Flight Mode voice alerts selection is dependent on the telemetry link.
This is fine most of the time. But occasionally it messes up at the edge of telemetry range or even reads incorrect if a variant of that flight mode has been chosen in iNav.
What I prefer to do, is just place the Flight Mode voice alerts in their switch positions via a Special function in OpenTX.

My request is - Would you be willing to add another Voice Alert options? 
 Like - ` ALL_non flight mode`

pkerney, 2019-09-17:
Adding a comment to the "edge of telemetry" part of this request. I was at the edge and the voice alert told me "Home Reset" which led me to think that I could not rely on RTH and crashed. Upon recovery, the BlackBox logs did not show any home reset or anything out of the ordinary so I am thinking that the telemetry was corrupt and therefore gave me incorrect feedback. This would be probably the opposite of what Jetrell is asking. Not sure what the solution/recommendation is here.

teckel12, 2019-09-17:
The solution to this is turning off all voice alerts (an option in Lua Telemetry). Or, fixing telemetry from the receiver manufactures so the protocol includes error detection. This isn't a Lua Telemetry issue.

closed 2019-09-17 by teckel12

Jetrell, 2019-09-17:
@teckel12 . Please understand that I'm not blaming Lua Telemetry for this.

And I understand that voice alerts can be turning it off is an options. But I don't necessarily what every voice alert off.
GPS/LQ/RFMD etc, are useful.
I just figured that it would a higher likely hood of you doing a work around for use. Than waiting on TBS.

Jetrell, 2019-09-17:
@teckel12 . Please understand that I'm not blaming Lua Telemetry for this.

And I understand that voice alerts can be turned off in the options. But I don't necessarily what every voice alert off.
GPS/LQ/RFMD etc, are useful.
I just figured that it would be a higher likely hood of you doing a work around for us. Than waiting on TBS.

Jetrell, 2019-09-17:
@pkerney I too have experienced a similar thing at the edge of telemetry range.

When doing longer range flights with planes, its easy to push passed the 25mW telemetry limit. 
Yet its great to have Lua Telemetry operation on the radio in the background, for when it comes back in range.  
The only draw back is that Amber gets confused from time to time.

teckel12, 2019-09-18:
@Jetrell There are lots of voice alerts that are not flight modes that could also cause confusion: arm/disarm, heading hold, headfree, alt hold, home reset, etc. And even if we consider those flight modes or flight mode variations, there are other voice alerts which could cause panic if there's telemetry corruption. For example, battery or telemetry critical or an altitude alert. 

After we remove all of those, there's really no more voice alerts left. Then, you may as well just turn off all voice alerts. Critical may be a good compromise, as it will still let you know when link quality or battery is critical, and the only flight modes it will announce will be RTH and failsafe. It looks like the only thing you may be missing is the battery percent voice alerts. 

The real solution here is that the telemetry protocol should have some type of error detection. If there's an error, throw out that telemetry. Maybe there's a checksum or some kind of simple error detection but it's not being used? (TBS would need to confirm this).

teckel12, 2019-09-18:
@Jetrell This wasn't mentioned here but voltage and link quality corruption can be "solved" by checking the "Filter" box for the telemetry sensors.

Also, make sure your using Lua Telemetry 1.7.2+ because there was a change in 1.7.2 having to do with limiting reporting of a possible invalid flight mode if telemetry is lost. So it's possible that in 1.7.2 there's fewer incorrect alerts at the fringe of reception.

Jetrell, 2019-09-18:
Thank you for the reply Tim,  I see what you mean. 
I will give your recommendations a try. :+1:

Jetrell, 2019-10-10:
@teckel12  I'm still having issues with the FM voice alerts and CRSF. 
And even with the FM display, showing `Manual` in some cases, while the mode switch has been changed to `Acro` mode etc.
The FC is in Acro, which can be seen from the servo stabilization.

I have taken your advice from earlier posts. And even set voice alerts to Critical. Which works. But I still like to hear the GPS audio alert... Which it does not speak when Critical alerts is selected.

I'm not blaming LT. But it would still be helpful to have the option of ALL voice alerts, except FM.

teckel12, 2019-10-11:
@Jetrell I believe the solution here is for Crossfire to not output corrupted sensor values. I don't think there's a fix for Lua Telemetry. No flight mode only alerts isn't an option. The solution it to fix it at the source of the problem, not a patch that removes functionality.