teckel12/LuaTelemetry #349: Support OpenTX's new virtual key events

teckel12, 2019-09-23:
### Is your feature request related to a problem? Please describe
OpenTX v2.3 adds new virtual key events that simplifies the controls of multiple controllers.

### Describe the solution you'd like
Clean up Lua Telemetry code to use the virtual key events

### Describe alternatives you've considered
Continue the painful way it's being done now, where every controller needs to be hand-built.

### Additional context
For Lua Telemetry to continue to work with older versions of OpenTX, detecting of missing virtual key events will be required.  Another option is to require OpenTX 2.3, but I don't think that can be done till a later date (but will be done at some point so legacy code can be removed).  Maybe 6 months after OpenTX 2.3 is released it's made a requirement of Lua Telemetry?

teckel12, 2019-09-30:
I'll wait till the beginning of 2020 to switch over to using the virtual key events.  Seems `EVT_VIRTUAL_MENU` is missing from the Jumper T12 and there could be other issues I'm not aware of at this time.  By January of 2020 it's not unrealistic to require OpenTX 2.3+

LawrenceLDJr, 2019-12-31:
Jumper T12, INav and OpenTX 2.3.4, new key config, naming, mapping, configuration of the T12s 6 buttons has a different instruction set in OpenTX. I imagine this is why only the enter key works with INav now. Sounds like a simple fix. Happy Holidays

jj-uk, 2020-02-19:
Is there a branch anywhere that contains the required changes for OpenTx 2.3.5  ?

teckel12, 2020-02-25:
@jj-uk There's not currently a branch with support for the new virtual key events.  It's more than just a simple search and replace.  The change will break anyone using OpenTX prior to v2.3 which is why it hasn't been changed yet.  Once the switch has happened, OpenTX 2.3 will be required.  This spring (as it's currently too cold to fly where I'm from).

jj-uk, 2020-04-01:
There's a fix in Open-tx 2.3.6 that addresses issues with the + - Exit buttons.  Not sure if this is related?
https://github.com/opentx/opentx/pull/7356