teckel12/LuaTelemetry #351: Artificial Horizon in Pilot View Rarely Updates

mattjsmithone, 2019-09-24:
The Artificial Horizon in pilot view updates very slowly on both the Qx7 and T16 transmitters (in the range of about once every 10-15 seconds). This means the feature isn't useful as it is almost always 'out of date' in relation to the aircraft's actual attitude. This seems unintentional as the hdg displays update several times a second. Is there anyway to solve this?  

Using Open TX 2.2.3 and Jumper TX 2.2.3 on the radios (two QX7's and one T16).
Running the latest versions of R9 firmware on all three, one of which is fcc, the others are flex.
The flight controllers are all running Inav 2.2.1 on a Matek F411-Mini, F405-Wing and F411-WSE.
Smartport Baud rate manually set to 57600 on one of the models also.

Love the software nevertheless,
Thanks in advance
Matt

teckel12, 2019-09-24:
@mattjsmithone I would like more information because on my Qx7 it updates at 20 frames per second (the max speed of lua scripts).  On the Jumper T16 it's slower due to the larger color screen, but it's about 8-10 frames per second.

My guess is that Lua Telemetry is running at the max speed possible on the Qx7.  The issue must be slow telemetry.  I don't fly R9.  I fly FrSky X-series or Crossfire and both are very fast (Crossfire does get slower at long range).

Is it possible that it's fast when you're close-range and only when long-range it gets very slow?  Again, that would indicate a limitation of the telemetry protocol.

I would like to get telemetry log files so I could duplicate your flight.  Instructions ca be found here:

https://github.com/iNavFlight/LuaTelemetry/wiki/Playback-Telemetry-Log-Files#setup

In any case, this is not an issue with the script, you have something else causing things to be slow.  The only other person who had a really slow screen was doing transmitter screen shots 10 times a second.  You're not doing something like that are you?

teckel12, 2019-09-24:
@mattjsmithone This video shows a very real-world example of the typical attitude update speed:

https://www.youtube.com/watch?v=YaUgywuT1YM

While Lua Telemetry is updating at 20fps, telemetry doesn't arrive quite that quickly (maybe 5 updates per second).  So the bottleneck is the telemetry protocol, not Lua Telemetry.

mattjsmithone, 2019-09-24:
Hi There @teckel12 ,

I have only installed this Lua in the last couple of days, so all of my testing has been 'on the bench' at close range. I could do a test and send you a log file along with a video of it? The earliest I could get a flight log would be at the weekend. I'm not taking any screenshots at all, so I don't expect that to be an issue.

Thanks for the fast response!

teckel12, 2019-09-24:
@mattjsmithone I can assure you, what you are experiencing is **not** normal and you're only the second person to ever report anything similar (and the other person had a function that took screen shots 10 times a second, which was to blame).

Are you doing a `set frsky_pitch_roll = ON` in INAV (and then doing another telemetry "discover new sensors") so the telemetry sensors `0430` and `0440` (`Ptch` and `Roll`) are created?  If not, it uses the raw accelerometer values which only give an attitude estimate and drift over time (not at all suggested).  `set frsky_pitch_roll = ON` is really required.

mattjsmithone, 2019-09-24:
@teckel12 Yes, done that! I renamed both 0430 and 0440 sensors to Ptch and Roll too. I'll create a log file and send it over shortly.
Thanks

mattjsmithone, 2019-09-24:
[BatWing-2019-09-24.csv.txt](https://github.com/iNavFlight/LuaTelemetry/files/3648538/BatWing-2019-09-24.csv.txt)
[VID_20190924_182204.mp4.txt](https://github.com/iNavFlight/LuaTelemetry/files/3648553/VID_20190924_182204.mp4.txt)
These are the log file and video, just remove the .txt extension for them to work

teckel12, 2019-09-24:
@mattjsmithone You have a problem with your telemetry.  It's not that Lua Telemetry is slow, it's that you're getting VERY slow telemetry values from your model.  Lua Telemetry is only reporting what it's getting.

It's also not just the roll and pitch, your other sensors are updating VERY slowly as well (like heading is only updating about once every 10 seconds).  Something seems seriously wrong with the connection between your flight controller and your receiver, your receiver, or your transmitter to be causing this.

To prove it's not Lua Telemetry, setup another page on your Qx7 that shows telemetry values.  Have it show `Hdg`, `Roll` & `Ptch`.  As you move your model, those should change quickly.  Looking at your telemetry logs, they will update about once every 10 seconds (they should be more like every 1/10 of a second updates).  It's seriously about 100 times slower than it should be, so something is seriously wrong.

I would also `Delete all sensors` and then `Discover new sensors` as you have sensors listed that are not being sent (like AccX, AccY, AccZ for example).

teckel12, 2019-09-29:
@mattjsmithone The following video shows the screen update speed.  I guess it was recorded upside-down and there's no way to rotate it on the phone or in YouTube, but it's just to show you the speed anyway:

https://youtu.be/N8NvaLe5Xb8

mattjsmithone, 2019-10-01:
Okay thanks for your help, must be an R9 problem then!

teckel12, 2019-10-01:
@mattjsmithone Fairly sure others are using R9 with Lua Telemetry.  Maybe update the firmware on your receivers.

closed 2019-11-22 by stale[bot]