teckel12/LuaTelemetry #352: Lua 1.7.2 speech

ds7060, 2019-09-24:
Hello,
I am using a Jumper T16 TX (FW 2.2.3 release 19-07-26) with a TBS CrossFire Micro module, and the iNAV Lua 1.7.2. 
I have a problem during the flight, the Lua start to speak "Manual Mode - Horizon Mode" and repeat this during all the flight ( on my OSD the mode is not changing ), then when I land and Lua start to speak "Engines Armed - Engines Disarmed". At this point, I notice something strange, when I go to pickup the plane on the runway, when I aproach the Lua stop speaking.
This is only a problem of speech because the mode is not changing and the motor is not arming/disarming. 
I don't know if this is a bug or not, a T16 issue, a Xfire problem or Lua 1.7.2 issue.
Thank you for the help. 
Regards.

teckel12, 2019-09-24:
Duplicate issue, see https://github.com/iNavFlight/LuaTelemetry/issues/315

Upgrade Crossfire to 2.95.  There's known telemetry issues with versions previous to 2.95.

ds7060, 2019-09-24:
I am running Xfire 3.21!!!

ds7060, 2019-09-24:
So please do not trash comments so quickly! Thank you

teckel12, 2019-09-24:
Lua Telemetry is just reporting what Crossfire is sending it.  Setup a widget screen that shows telemetry values.  Have it show the `FM` sensor value.  Lua Telemetry will match what Crossfire is sending.

It's also possible that new bugs have been introduced in 3.21.  I'm running 2.95 without any issues.

I'm not trashing your comments, the issue linked is the exact problem you're reporting with a confirmed solution.

teckel12, 2019-09-24:
@ds7060 Also, it's a known issue that the Crossfire protocol must not have good error detection with telemetry because it will sometimes send bad telemetry.  This typically only happens long-range, which would be why it would work when you're closer to your model and then not work when you're further away.  Again, showing the `FM` sensor value would allow you to see that this bogus information is coming from Crossfire and Lua Telemetry is just reporting it.

But, it does sound a little extreme in your case, could be they've introduced another telemetry bug in 3.21 like they did around 2.88.

Hope this helps!

teckel12, 2019-09-24:
@ds7060 Actually, I'm running 3.00 without any issues.  Haven't tried 3.21 yet so that could be the problem.

ds7060, 2019-09-25:
Thank you, and sorry I haven't seen issue #315

wx4cb, 2019-10-26:
works fine on 3.23 which is the latest release as of 9/26. i'm not using 2.3 on my flight controller, i think i'm still on 2.91 of inav itself

closed 2019-12-18 by stale[bot]

Eg681, 2021-04-17:
My setup is inav, crossfire 3.78. When plane gets out around 1/2 mile I start getting voice messages that its in position hold mode but I haven't switched modes, so is this a crossfire problem and not inav lua?

wx4cb, 2021-04-18:
> My setup is inav, crossfire 3.78. When plane gets out around 1/2 mile I start getting voice messages that its in position hold mode but I haven't switched modes, so is this a crossfire problem and not inav lua?

get off of 3.78 and install 4.11 on your crossfire module.. 3.7.8 is way old

Eg681, 2021-04-18:
I plan to right away, do you think that is my flight mode issue while in flight?

wx4cb, 2021-04-18:
> I plan to right away, do you think that is my flight mode issue while in flight?

there were issues with 3.7.8 and i've been running 4.11 for a couple months and don't have any issues.

Eg681, 2021-04-18:
Ok good news I'll give it a try, Thanks