teckel12/LuaTelemetry #353: Disable fuel graph

skorokithakis, 2019-10-13:
My controller doesn't have a current sensor, so the fuel bar is not very useful (I use the voltage to estimate battery level). Is there a way to disable the fuel bar, and could this be added to the Wiki somewhere in the configuration section?

teckel12, 2019-10-15:
@skorokithakis What transmitter do you have? I know on Taranis it will remove the fuel info if there's no fuel sensor.

teckel12, 2019-10-15:
@skorokithakis INAV may create a  fake fuel sensor if one doesn't exist. If you have a Fuel sensor on your transmitter, try deleting it and then run LuaTelemetry.

skorokithakis, 2019-10-15:
@teckel12 I do have a Taranis, I was reasonably sure I could delete the fuel sensor but wasn't entirely sure this wouldn't mess up something else, that's why I'm proposing it be added to the wiki.

I'll delete my sensor now, thanks!

teckel12, 2019-10-15:
@skorokithakis Yes, the script will allow you to remove the fuel sensor and it won't show. INAV added the Fuel telemetry even when there isn't a current sensor. This can be beneficial as they use an algorithm which should be useful and somewhat different than just the battery voltage.

But, if you don't want that, I believe you can even turn it off in INAV. Not really sure if anything needs to be said in the Wiki, I already mention that a current sensor is optional.

skorokithakis, 2019-10-15:
> I believe you can even turn it off in INAV.

Oh hmm, I didn't find how, maybe it's somewhere in the battery settings. Removing the sensor worked fine.

> Not really sure if anything needs to be said in the Wiki, I already mention that a current sensor is optional.

I didn't find that (or didn't understand it means I can just remove it). Anyway, if you think things are clear enough, that's fine, I'm just saying that I was confused on this.

teckel12, 2019-10-16:
@skorokithakis What was confusing about showing a fuel sensor? My guess is if you turn off fuel percent in INAV it won't set the sensor. All my flight controllers have current sensors, so I'm honestly not sure.

skorokithakis, 2019-10-16:
The confusing part was that I wasn't sure if I can just delete the telemetry sensor to make the fuel graph not show, or if that would break something, and the wiki didn't clarify adequately that it wouldn't break anything.

teckel12, 2019-10-16:
The current sensor is optional and mentioned. I'll see if there's a good place to clarify this.

teckel12, 2019-10-16:
@skorokithakis https://github.com/iNavFlight/LuaTelemetry/wiki/Installation#add-telemetry-sensors-to-transmitter

closed 2019-10-16 by teckel12

skorokithakis, 2019-10-16:
That's perfect, thanks!