teckel12/LuaTelemetry #356: Syntax error and corrupt SD Card

Cinntax, 2019-10-22:
### Describe your question
X9D+, iNav 2.2.1, OpenTX 2.3.1
I've installed this Lua script (the lightweight taranis-EN one) on the SD card by mounting the card in my laptop and manually copying the files over.
I have properly configured my model to use the iNav telemetry screen.
When i run the telemetry screen though, i get a syntax error (expected symbol near '.' inav.luac:1), and the /SCRIPTS/TELEMETRY/iNav folder is totally hosed on the SD Card (it wasn't that way going in!).
I suspect this may be something going on when compilation occurs, but i'm kind of at a loss at the moment.  I'm running the stock 1GB SD card.

Has anyone else seen this?  Didn't want to file a bug, as it could totally be my stupidity.  Thanks!

[image]

Cinntax, 2019-10-23:
Ok for future reference, my issue was a bad SD card. Evidently there was a batch of x9d’s that includes knock off sd cards that were ~170mb instead of 1GB. There is a nice write up about this here: https://intofpv.com/t-warning-frsky-transmitters-containing-fake-capacity-sd-cards
I swapped my sd card for a real one. Problem solved.

closed 2019-10-23 by Cinntax

teckel12, 2019-10-24:
Nice! I was like, "what"? with this issue.