teckel12/LuaTelemetry #358: Question on Jumper T16, crossfire, pixhawk not working

spyder0069, 2019-11-30:
I have a pixhawk that is setup to output mavlink through the crossfire diversity receiver, which works fine outputting through bluetooth to mission planner.  I am trying to get this to work with the telemetry script on the jumper t16.  If I go into the telemetry menu and search for sensors I get things such as:
rssi
crossfire power level
link quality
pitch 
roll
yaw
rxbattery (flight battery)
current sensor
ma used

I don't see any gps info coming in as a telemetry sensor.  In the script it displays the volts/current/rssi ok but I do not get any movement on the upper left display (attitude?) when i move the quad around even though the pitch roll yaw sensors in the menu do show movement.

Is this just not a supported setup yet or do I have a configuration wrong?

I am on crossfire 3.23 firmware.
Receiver menu is setup with mavlink in/out on the pins connected to telemetry 1 on the pixhawk.
Pixhawk has telemetry set to output in Mavlink1 mode at 57600
T16 set to crsf mode running opentx 2.3.2 (11/07/2019)

I appreciate any help I can get.  :^)

spyder0069, 2019-12-01:
After deleting and rescanning my sensors I now have all the data in my telemetry screen in the opentx menu including GPS info and pitch/roll/yaw.  The inav widget now shows the lattitude / longitude ok.  

It does not show the satelite count or hdop properly.

The battery in the lower left shows 0.0V but the flight battery is showing in top right in small text.

Something is funky with the compass stuff.  I think its trying to show heading on the three screens.  They are showing north but my quad is pointed south.  I'm indoors on a rainy day so those readout on the inav are moving around pretty good however mission planners nav is showing correctly south so it must be using the correct compass reading.

On inav altitude is showing the actual above sea level altitude.  However, mission planner and my osd to tar it out at zero altitude.

Pitch/roll are not updated in the inav display (show properly on mission planner).

teckel12, 2019-12-07:
@spyder0069 You're running an unsupported configuration. LuaTelemetry is for INAV only. Well, it works a bit with Betaflight, but with limited support.

closed 2019-12-07 by teckel12