teckel12/LuaTelemetry #360: alternating views on jumper t12 its possible?

nangel0, 2019-12-05:
is alternating views on jumper t12 its possible? pressing the left and right buttons on default view does nothing on jumper t12

### To reproduce
latest opentx on jumper t12
r9m telemetry enabled
default view on inav lua script works , others doesnt

teckel12, 2019-12-07:
@nangel0 you can change views from the config menu. You probably can't change views while in a view due to non-standardization of controls from transmitter to transmitter. This will be resolved when OpenTX merges support for Jumper/Nirvana transmitters in the main source code.

closed 2019-12-07 by teckel12

PasiRuo, 2019-12-08:
Latest LuaTelemetry worked for one period of OTX2.3.0nightlies.

Virtualbuttons have since changed. I can change view with shortENT, but not possible to enter config-menu. I found nothing to still be OTXmerged regarding improvement of buttons for T12

PasiRuo, 2019-12-10:
to put in other words, T12 passes virtual button test ok
https://github.com/opentx/opentx/blob/2.3/radio/util/lua/lua_event_test.lua
but LuaTelemetry does not open config with virtual_menu