teckel12/LuaTelemetry #362: Different screen on my horus x10

fmartinezo, 2019-12-14:
Hello i have different screen on my horus that the screen wich appears on github see my photo specially on top left panel and compare whith your github panel.
What is wrong on my radio?
[image]
[image]

teckel12, 2020-01-23:
Are you flying FlySky or Crossfire?  If you're flying FlySky, is seems you don't have `frsky_pitch_roll` set to `ON` as required.  Please see the installation instructions:

https://github.com/iNavFlight/LuaTelemetry/wiki/Installation#inav-configurator-telemetry-setup

fmartinezo, 2020-01-23:
I am using frsky x8r and rsxr. The problem was i do not read the 3rd point
of change cli parameter pitch-roll. As Soon is i Will arrive at home i try
It
Thanks

fmartinezo, 2020-01-25:
Ready, The problem was that set frsky_pitch_roll = ON and rediscover new sensors was new for me. All is working fine. Thanks

closed 2020-01-25 by fmartinezo