teckel12/LuaTelemetry #363: INAV Telemetry - Battery critical warning and Fuel % showing.

simtcrom, 2019-12-15:
### Describe your question
Even though I have full battery, Taranis telling battery critical and battery 0 %

### To reproduce

1. Go to '....'
2. Click on '....'
3. Scroll down to '....'
4. Should it be doing this?

### Expected behavior

### Screenshots

### Additional context

simtcrom, 2019-12-16:
This issue is resolved, I had mistakenly rename telemetry sensor for Fuel to Mah.
Deleted all sensors and rediscovered again, all issues resolved

simtcrom, 2019-12-16:
This issue is resolved, I had mistakenly rename telemetry sensor for Fuel to Mah.
Deleted all sensors and rediscovered again, all issues resolved.

closed 2019-12-16 by simtcrom

darvin, 2020-02-07:
I have the same issue, removing sensors and rediscovering of them doesn't help

iabdullah98, 2023-04-15:
i have the same problem and it didnt solve it