teckel12/LuaTelemetry #365: Ability to select models lost after running InavFlight

robthomson, 2019-12-20:
On the latest RC2 candidate of opentx; on the jumper t16...

all works well.

However.. if you then try select a new model; the model names have all been wiped out on the model select screen.

It is a bit like the script is over-writing settings on the main tx page.

teckel12, 2020-01-23:
@robthomson Selecting a model has nothing to do with LuaTelemetry.  Maybe a screenshot would explain things better as it sounds like an OpenTX bug to me.

ThaBoom, 2020-02-18:
I have found the same issue. Playing around I found that booting up with a model that has the script enabled causes this issue.  Also makes the last shutdown screen (the sleepy sun)  have a border.  Maybe a background color is not being unset?  I am running opentx 2.3.5.  If you need testing I can do it.

[image]

robthomson, 2020-02-19:
I eventually sorted mine by changing to a different theme.

The default one has this issue.. The blue one.. Seems OK.

ThaBoom, 2020-02-19:
I found the fix it seems to be this.

https://github.com/iNavFlight/LuaTelemetry/issues/333#issuecomment-511234468

teckel12, 2020-02-19:
Ah yes! If you're using a light-colored background theme this will happen due to the way widgets work. The solution is to use either a dark-colored background theme (recommended) or in the widget setting for Lua Telemetry set the colors you want to restore for the main font color and warning color.

closed 2020-02-19 by teckel12