teckel12/LuaTelemetry #366: AltHold Haptic Feedback

namirda, 2019-12-26:
The "AltHold Center FB - Hepatic/audio" feature works great when you take-off with AltHold OFF and then subsequently switch it ON during the flight. 

However, it does not seem to work when you take-off with AltHold ON - no haptic or audio feedback received until you exit AltHold mode and then enter it again..

The INav parameters relevant to AltHold for my model are as follows:

nav_use_midthr_for_althold = OFF
nav_mc_hover_thr=1350
thr_mid=35
alt_hold_deadband=50

It's a minor problem - am I doing something wrong here?

Thanks

namirda, 2019-12-27:
I have just taken a look into the code and I think the problem is at line 384 of iNav.lua 

data.thrCntr = data.throttle

which sets the thrCntr to be the throttle whenever AltHold is engaged. 

In the case of a take-off with AltHold engaged, thrCntr should instead be set to the value of the iNav parameter "thr_mid". Unfortunately I don't think this value is available to the script.

One solution is to add another parameter to the LuaTelemetry config which the user should ensure contains a copy of "thr_mid"??

This seems like a lot of work for a minor problem - much simpler to just add an explanatory note to the documentation saying that this AltHold feedback only works if altHold is engaged during flight!

Thanks

teckel12, 2020-01-23:
@namirda  You are correct, this function is only works if you activate alt hold while in the air and you have your have nav_use_midthr_for_althold = OFF.  I've looked into the solution you mention, but there's no way to correctly get the throttle position due to different rates being set.  Even if you could set the value to 1350 in LuaTelemetry, that's the value that INAV is receiving, not necessarily what the actual throttle position is.

Therefore, this feature unfortunately will require alt hold to be activated while in the air, not on the ground (which I've never even heard of till now).

closed 2020-01-23 by teckel12