teckel12/LuaTelemetry #371: no sounds except acro mode and horizon

christos139, 2020-01-15:
i have update firmware in my taranis qx7 2.3.4 and i have no sounds in lua script except horizon angle and acro mode...before i have 2.2.4and everything was fine ... its this a bug or i have to do something else i cant figure out?

teckel12, 2020-01-23:
OpenTX v2.3.x has not yet been tested as it's winter and therefore too cold to fly (for me).  It's possible that 2.3.4 breaks something.  It would be nice to know in what revision between 2.2.4 and 2.3.4 something stopped working for you.  But at this time I would use 2.2.4 till it warms up at least.

christos139, 2020-01-23:
at the moment because i havent fly me too the only thing i saw was the sounds....i cant hear sounds ,like manual rth and such things...only when i get some gps satellites some of them start "talking" again

teckel12, 2020-02-19:
@christos139 If there's no satellites maybe it's not saying anything because it's in a "not ready to arm" state. You really need a sat lock for all the modes to function, or it's in a "GPS Lost" state.

Multisoft1978, 2021-03-19:
I am using Qczek with LoRa modules
I used to have firmware 2.2.3 or 2.2.4, I don't remember, everything worked fine. Now I have changed the firmware to 2.3.11 and I only hear the engines on, acro mode. On the screen, I see acro mode all the time.
I rolled back to 2.3.9 and still only acro mode.
I rolled back to 2.2.4 and again only acro mode.
I don't know what happened and what to do now.

teckel12, 2021-03-19:
@Multisoft1978 Maybe try to delete all telemetry sensors and let the transmitter re-add them.  I've seen a few cases where people rename the sensors which will then no longer work as it's expecting the default names.  If you're still having a problem, look at the raw sensor values and see what changes when you arm and change flight modes (specifically, the `Tmp1` and `Tmp2` flight sensors).  Which, by the way, need to named exactly that, and not changed in any way (like every other sensor).  The only setting you can change is the unit of measure, and only for the speed/distance sensors.

Multisoft1978, 2021-03-19:
> @Multisoft1978 Maybe try to delete all telemetry sensors and let the transmitter re-add them. I've seen a few cases where people rename the sensors which will then no longer work as it's expecting [...]

I tried different firmwares and constantly updated, removed and added sensors.  Tmp1 and Tmp2 are there, when I switch the toggle switches, the data changes, but ACRO is constantly on.  If in Tmp1 change the temperature of Celsius (C) to Fahrenheit (F) then the data randomly changes in the script on the display, without toggle switch.

teckel12, 2021-03-19:
Yeah, don't change the unit of measure for Tmp1 and Tmp2, that's how INAV sends the flight modes.  Any changes at all to those sensors and nothing will work, they're not temperatures, they're a binary representation of the flight modes.

I'd need to know the exact values of Tmp1 and Tmp2 to diagnose this.  Either recording your telemetry and adding it to this issue or at least noting what the values are at different times, for example:

1. Before armed, Tmp1 & Tmp2 values
2. After armed, Tmp1 & Tmp2 values
3. Change flight mode, Tmp1 & Tmp2 values

You can even do this without flying, just being armed and changing flight modes.  Obviously, telemetry logs would be the most useful, but you'd need to set your transmitter to record telemetry.  I always do this as it's a good way to monitor anything that happens, and you can also playback a flight on the transmitter with LuaTelemetry.

The following shows instructions on how to record telemetry logs: https://github.com/teckel12/LuaTelemetry/wiki/Playback-Telemetry-Log-Files

Multisoft1978, 2021-03-20:
> Yeah, don't change the unit of measure for Tmp1 and Tmp2, that's how INAV sends the flight modes. Any changes at all to those sensors and nothing will work, they're not temperatures, they're a [...]

https://drive.google.com/file/d/1mXCltDhLMAjLOhcIiU7TFO3lpmnBQVCj/view?usp=sharing
https://www.youtube.com/watch?v=8nuXV9KAXfY
https://www.youtube.com/watch?v=sHaZX7VP5v4

teckel12, 2021-03-22:
@Multisoft1978 You have something strange going on.  Your Tmp1 and Tmp2 sensors are all messed up.  The video shows that Tmp1 just changes from 1 to 4 and that's it.  If you're changing flight modes, that would cause Tmp1 to change as well.

Also, your csv flight log shows not only Tmp1, but also 440, 04A0, 04A1, etc.  It's like your sensors have been customized.  You should delete ALL telemetry sensors and then have it search for telemetry sensors.  This will add them all back as factory defaults.

At a minimum, delete Tmp1, Tmp2, 400, 04A0, 04A1, 04A2, 04B1, 04B2, 04B3, 430, 440 and then search for sensors.

I'd also suggest you do a `set frsky_pitch_roll = ON` in INAV as per the install instructions.  I'd also go over the install instructions as it seems you've changed settings which is causing the problem.  The install instructions are here:

https://github.com/teckel12/LuaTelemetry/wiki/Installation

closed 2021-03-22 by teckel12

Multisoft1978, 2021-03-23:
Hello, I add set frsky_pitch_roll = ON
But this did not solve the problem.
Deleting all the sensors and searching for them gives the same result.
https://youtu.be/nvOHQH-IC5M
I do not know what the problem may be, I have already used two different flight controllers and two LoRa modules. I tried different INAV firmware from 2.1.0 to 2.6.1 I flashed Taranis Q X7 from 2.2.4 to 2.3.11 and the result does not change.

Sensors QCZEK https://qczek.beyondrc.com/qczek-lrs-433mhz-1w-lora-rc-link/qczek-lrs-sport-telemetry/
APID_GPS_COURSE 0x0840
APID_RSSI 0xF101
APID_VFAS 0x0210
APID_CURRENT 0x0200
APID_CELLS 0x0300
APID_ALTITUDE 0x0100
APID_VARIO 0x0110
APID_GPS_SPPED 0x0830
APID_LATLONG 0x0800
APID_GPS_ALTITUDE 0x0820
APID_AIR_SPEED 0x0a00
APID_FUEL 0x0600
APID_T1 0x0400
APID_T2 0x0410
APID_PITCH 0x0430
APID_ROLL 0x0440
APID_MAV_BASE_MODE 0x04A0
APID_MAV_SYS_STATUS 0x04A1
APID_MAV_CUSTOM_MODE 0x04A2
APID_CUST_RSSI 0x04B0
APID_RX_RSSI_REG_VAL 0x04B1 – with offset 157 for 868MHz and 146 for 433MHz
APID_RX_SNR_REG_VAL 0x04B2 – with offset 64.
APID_RX_PACKET_LOST_VAL 0x04B3

Multisoft1978, 2021-03-23:
> @Multisoft1978
I was told that the MavLink protocol for this will only be Acro. Is it so? But how did it work for me before and for other people it works ?!