teckel12/LuaTelemetry #372: Mapping Widget like in Yaapu Frsky Telemetry Script to show where plane in on a map

Gaylian, 2020-01-21:
Mapping Widget like in Yaapu Telemetry Script project, a LUA telemetry script for the Frsky Horus and Taranis radios using the ArduPilot frsky passthru telemetry protocol.

teckel12, 2020-01-23:
@Gaylian This already exists, see https://github.com/iNavFlight/LuaTelemetry/blob/master/assets/iNavQX7radar.png

closed 2020-01-23 by teckel12

Gaylian, 2020-01-24:
Hi

How do I get the maps to show like this with INAV.

 

Closed #372 <https://github.com/iNavFlight/LuaTelemetry/issues/372> .

teckel12, 2020-01-24:
@Gaylian Sorry, on the T16 it looks just like on a Horus:

[image]

All you need to do is follow the installation instructions: https://github.com/iNavFlight/LuaTelemetry/wiki/Installation

I'm a little perplexed as to what you're question is.  If you want to know how to use LuaTelemetry, you just need to install it via the instructions.  There's a bunch of other Wiki documents if you have other questions: https://github.com/iNavFlight/LuaTelemetry/wiki

Gaylian, 2020-01-25:
Hi Tim

I have no problems getting telemetry to work.

What I was after was superimposing the position of the plane over a topological map.

 

So it will look like this. Its not the telemetry info that is my problem but getting the plane pointer to appear on a map as in this pic.

 

 

 

Regards 

 

Adrian Brown

 

 <https://travis-ci.com/iNavFlight/LuaTelemetry>