teckel12/LuaTelemetry #373: Script syntax error - Taranis Qx7s

ilovefpv, 2020-02-01:
Hello, I have 3 models on Inav and 3 supposed to use Inav lua script. Two of them were working fine, the third one too until today for an unknown reason as I did not change anything a part of the voltage scale in inav (matek f722 change from 110 to 183 - voltage info was wrong).

Now I have "script syntax error / bad argument #1 to 'min'..." 

I tried re discover sensor etc...but no change.

Please help

[image]

teckel12, 2020-02-19:
@ilovefpv Not sure, but this may help me help you:
 https://github.com/iNavFlight/LuaTelemetry/wiki/Issue Template

teckel12, 2020-02-25:
@ilovefpv Downgrade to OpenTX prior to 2.3.  2.3+ is not yet supported by Lua Telemetry due to breaking changes.

closed 2020-02-25 by teckel12