teckel12/LuaTelemetry #374: Lua buttons weird behaviour on X9D+ OpenTX 2.3.5

Mateyhv, 2020-02-03:
Hi, having problems navigating in iNav LuaTelemetry 1.7.2 on a Taranis X9D+ running latest OpenTX 2.3.5 so the issue can be related to this version either, not really sure since I have not used this lua for several months. 

Description: Everything is set as per the installation manual. 
1) Once into the lua page if I have set other OpenTX telemetry pages, pressing '+' and '-' buttons instead of changing the lua values, jumps to the next telemetry page (not lua page), same behavour as if 'page' button is pressed. If I delete all telemetry pages, then '+' and '-' buttons behave correctly.
2) Once into the lua page if 'menu' is pressed the lua menu displays. If there are other telemetry pages '+' and '-' behave as in point 1. If not, the buttons act normaly. Regardless of other telemetry pages when pressed 'exit' nothing is saved and the screen returns to the main display page instead of the lua page. Pressing 'page' goes straight to the lua menu page, so there is no way to return to the lua screens unless the Taranis X9D+ is rebooted and lua settings are not saved.

I tested with another model wich was working properly a few months ago (with a now way older OTX version) and had the same result. Reinstalled freshly downloaded lua archives 1.7.2, same result.

Thanks

jj-uk, 2020-02-15:
I also have these issues. Once the LUA menu is entered, it cannot be exited back to "classic"view, and settings are not saved.
There is no way to exit the inav LUA menu without exiting the script.

Requires the radio to be power-cycled to get back to the "classic" view.

marcodcribeiro, 2020-02-15:
I have the same issue in the Taranis X9D and the cordenates show 0.0000000 with 16 sat

teckel12, 2020-02-19:
This has to do with OpenTX changing the button mappings. There's no good way to make it work consistently with pre OpenTX 2.3 and post 2.3. If I make it work for 2.3+ it will be totally non-functional for pre 2.3. Basically, OpenTX screwed up by not making things backward-compatible. They also introduced a few bugs that makes missed some key events.

This is a known issue, and the solution has been to wait till 2.3 matures, fixes the missing events, and then Lua Telemetry will be updated to use only the new key events. When this happens, versions of OpenTX prior to 2.3 will no longer work.

As there's a lot of people using pre 2.3, I figured I'd give it some time before forcing it. An update this spring will address this.

teckel12, 2020-02-19:
See https://github.com/iNavFlight/LuaTelemetry/issues/349

closed 2020-02-19 by teckel12