teckel12/LuaTelemetry #378: AHI jumping around

wx4cb, 2020-04-17:
### Describe the bug
Transmitter T16 (also happens with X9D+), when i go into the LUA script the AHI in the top left is always jumping around, even when the aircraft is standing still. not sure if it's a setting i need to change or not.

CLI Dump (diff) 

[INAV_cli_WX4CB-Harrier_20200417_155042.txt](https://github.com/iNavFlight/LuaTelemetry/files/4494954/INAV_cli_WX4CB-Harrier_20200417_155042.txt)

See attached video: https://youtu.be/gjXLpJQOQDM

Model name is "Harrier"

[Harrier.zip](https://github.com/iNavFlight/LuaTelemetry/files/4494946/Harrier.zip)

teckel12, 2020-07-07:
@wx4cb View the telemetry values and see what the `AccX`, `AccY`, `AccZ` values are doing.

wx4cb, 2020-07-07:
will do when i go out tomorrow.. welcome back btw 👯

closed 2020-08-29 by stale[bot]

wx4cb, 2021-02-16:
ok, just remembered about this..... i found out the problem.... it's because i have the sensor values aset as degrees (cus i know degrees not radians :D) 

must be a conversion thing you're doing in the script