teckel12/LuaTelemetry #384: Menu Issues

mrduncan626, 2020-05-07:
I’m currently using this LUA on an FrSky Taranis Plus configured with OpenTX 2.3.5.  I’ve been using this LUA for several years with no issues but recently, the menu screen is not working correctly.  When I’m in the menu screen, every time I move the up / down the menu or try to change a parameter value, the radio pops up of the menu screen back to the Taranis telemetry screen.  When I go back into the menu screen, the change was accepted but I have to go back and forth between the menu screen and the telemetry screen to make any changes.  However, when I’m done with the changes, the save option is not available so all the changes not saved.  And, the only way to get back to the LUA screen is to recycling power on the radio.

machadofelipe, 2020-05-21:
Same here. I have a old X9D.
If I enter in the Menu, I cannot go out without recycle power. I only detect the issue in the latest version.

Mateyhv, 2020-06-08:
> 
> 
> I’m currently using this LUA on an FrSky Taranis X9D Plus configured with OpenTX 2.3.5. I’ve been using this LUA for several years with no issues but recently, the menu screen is not working correctly. When I’m in the menu screen, every time I move the up / down the menu or try to change a parameter value, the radio pops up of the menu screen back to the Taranis telemetry screen. When I go back into the menu screen, the change was accepted but I have to go back and forth between the menu screen and the telemetry screen to make any changes. However, when I’m done with the changes, the save option is not available so all the changes not saved. And, the only way to get back to the LUA screen is to recycling power on the radio.

That was an odd behaviour of  OTX 2.3.5. With current 2.3.7 everything is back to normal so please close the issue.

machadofelipe, 2020-06-08:
> That was an odd behaviour of OTX 2.3.5. With current 2.3.7 everything is back to normal so please close the issue.

Agree. I have noticed the same as I updated for 2.3.7 the other day. Thanks :)

@mrduncan626, can you close?

teckel12, 2020-07-07:
Issue was with OTX 2.3.5, fixed with 2.3.7

closed 2020-07-07 by teckel12