teckel12/LuaTelemetry #386: Repeated voice messages - bug reintroduced by crsf?

HerpDerpula, 2020-06-07:
### Describe the bug
Exactly the same as a bug posted previously (I seem to have lost the #) where the radio will repeat a long loop of voice messages such as altitude hold on, heading hold off, heading hold on, etc. Mostly related to modes. Sometimes on landing it will think the engine was armed when the craft is configured to be disarmed. 

Jumper t16 pro v2 
OpenTX 2.3.7
Lua telemetry 1.7.2
iNav 2.4.0 on Matek405SE - wing
Crossfire firmware 3.28

### To reproduce
1. Arm craft.
2. Take off.
3. Fly a reasonable distance away, but still with 100% Lqly
4. Various voice tones repeat

### Expected behavior
Determine whether or not this is a configuration or crsf issue and try to resolve this.

### Screenshots

### Radio and model settings

### Additional context

HerpDerpula, 2020-06-07:
Is there a procedure for forwarding issues like this to crossfire? It appears your code is actually just fine and instead there needs to be some more robust telemetry error checking from TBS instead.

wx4cb, 2020-07-03:
no clue. the only way to do it i beleive would be to create a ticket. i'm not sure where tim is (i hope he's ok) but i can confirm it's still an issue.

teckel12, 2020-07-07:
I've seen this with Crossfire as well.  Seems to happen when long-range and the signal is weak (the entire point of Crossfire, right?).  Anyway, the flight mode freaks out and jumps between what it really is and something else (or maybe nothing?).

This is a garbage in/garbage out type of problem.  As I recall, I did make a change to limit this a bit.  You may want to try running the [development branch of LuaTelemetry (v1.7.3)](https://github.com/iNavFlight/LuaTelemetry/wiki/Upgrade-to-Development-Build) to see if that makes things better.

wx4cb, 2020-07-08:
the current dev version on the wiki works well for the most part (the only version i run), the only issue i've repeatedly seen is that when you land and disarm, when you pull the battery out of the module while the TX is still on it thinks you have lost signal while you are armed and it goes into telemetry lost. not really that big a deal for me as usually i turn off the TX anyway at that point

HerpDerpula, 2020-07-08:
Concur with the "lost telemetry" message occurring and that it's not a major issue. Mostly bothered by in flight issues, especially on those long range flights where the anxiety is high. 

It seems to be a bit better lately, which may be due to updates to the crossfire software. I'll try out the dev branch and report back. From there, I'll likely open a bug with crossfire and ask others to do the same to encourage them to fix this issue.

teckel12, 2020-07-08:
> the current dev version on the wiki works well for the most part (the only version i run), the only issue i've repeatedly seen is that when you land and disarm, when you pull the battery out of the [...]

I believe this is what Crossfire is actually sending to the transmitter, which is sent to LuaTelemetry.  Instead of just losing telemetry like Frsky receivers, Crossfire changes the flight mode and then loses telemetry.  As a result, LuaTelemetry is reporting what it's being told, which isn't technically right.  I've suggested to TBS to not do that.  Instead, keep the flight mode as-is and just have the telemetry signal drop, as that's what's really happening.

teckel12, 2020-07-08:
To fix this, Crossfire needs to report the last good flight mode instead of defaulting to no flight mode when telemetry cuts out for a second (or when the battery is disconnected).

Xedos9er, 2020-07-24:
Have you tried "force telemetry" mode in your Crossfire settings or switching off "dynamic power".

wx4cb, 2020-07-24:
> Same issue here, but also with dev build.

Doesnt matter

elbartnox, 2020-08-02:
I also have the same problems. It seems to happen when the rfmode or the LQ changes.
I have continuous flight mode change voice messages, which make the FPV script useless.
My crossfire is updated to ver.3.28, inav 2.3, Taranis X9D with Full crossfire module. Lua script 1.72.
I hope for a solution .. if it exists

HerpDerpula, 2020-09-20:
still occurs, keeping active until crossfire fixes this to prevent duplicate issues being written

Xedos9er, 2020-10-09:
I´ve created a support ticket on TBS mentioning this topic

sirPerna, 2020-10-13:
We just fixed this issue inside Crossfire. Sorry this took so long. It will be in the next release (>=4.06) which should be released within the next week.

Xedos9er, 2020-10-13:
@sirPerna 
Thanks a lot! I received a message from "Mike Chin" that it's an issue with opentx and getting fixed on 2.4. at some point that made no sense to me. Glad that you sorted it out.

Xedos9er, 2020-10-29:
@sirPerna 
I just installed the Update. It introduces a new bug. I get random "Telemetry lost, Telemetry recovered" massages and can see that the receiver goes randomly into failsafe. 
Going back to 4.05 solves the issue. So for now 4.06 is absolutely unusable.

sirPerna, 2020-10-30:
@Xedos9er this is very unfortunate. We have not received similar feedback from other users. Can you give me more details. Maybe send it directly to me via sir.perna@gmail.com

Xedos9er, 2020-11-20:
FYI, TBS finally fixed the weird flight mode issue on 4.08. I´m testing it for a couple of time now and it works without any issues.

Is there any development going on on this script??

wx4cb, 2020-11-20:
> FYI, TBS finally fixed the weird flight mode issue on 4.08. I´m testing it for a couple of time now and it works without any issues.

That's good to know

> Is there any development going on on this script??
i'm not sure... @teckel12 was very active up until the latter part of this year when it looked like he just fell off the face of the planet. i havent seen any posts from him or updates (which is unusual to say the least) for quite a while now. i personally am praying for him and that he's ok.

closed 2021-02-16 by teckel12