teckel12/LuaTelemetry #387: google maps overlay

purpletreee, 2020-06-08:
First of all thank you for the awesome script. It is working perfectly on my Radiomaster TX16s / Jumper t16. 
One feature which I really would like to see is a second screen with a satelite based map to see where the model is. This would be a perfect extension for those new radios with big and colorful screens. 
I just tried doing it on my own by changing the background image of the map to a screenshot from gm, but I think this won't do the trick. And it is really small also. 

Please keep this awesome project alive ;)

HerpDerpula, 2020-06-08:
Google maps will likely require some sort of business account. An open source alternative is preferred. You can access landsat data via API but don't expect landmarks to be labelled.

A framework to handle geotiff's will need to be implemented. 

https://www.usgs.gov/land-resources/nli/landsat/landsat-data-access?qt-science_support_page_related_con=0#qt-science_support_page_related_con

The two big things that stick out to me are:
1) how does the user pick the data for their location? It'd likely need to be manual.
2) how much resources from the radio is the LuaTelemetry team willing to dedicate to this capability?

Neat idea, nonetheless. Would help locate aircraft.

wx4cb, 2020-07-03:
take a look at mwp tools. there is no way it's going to work on the lua script. you would need an internet connection, and at least massive memory increase in the transmitter. it's just not physically possible with the radios we have. your best bet is to use a ground station computer/laptop which is what I do

teckel12, 2020-07-07:
Super cool request, and one I've thought of.  But there's zero chance of it working in a Lua Script.  There's just not enough memory/resources to do anything that complicated.

closed 2020-07-07 by teckel12