teckel12/LuaTelemetry #389: script syntax error bad argument #1 to 'min"

JupaCreations, 2020-06-11:
### Describe your question
<X7 with Inav lua telemetry script added results in above error after a while.
After the first time installing the script it worked fine and I could use it.
Suddenly out of the blue after a few flying days when you enter the LUA script screen I get this message.
New model made with same result.
Took a new SD and copied saved old model settings without LUA script added to the SD.
Old settings working fine for models.
Added the LUA script again on the card and set up one model with the script for test.
It works fine.
Suddenly out of the blue it stops working again with same error after a few days flying..
No models added or whatever which can cause memory shortage.
Philips 8GB class 4 SD card.
[image]
[image]

### To reproduce
1. Press PAGE button

### Expected behavior
<Show telemetry screen>

### Screenshots
<attached>

Strange problem which not seem a bug as I cant find the error message with google.

Paco

zsdanlli, 2020-06-19:
I got the same problem yesterday, fixed by deleting the <modelname>.dat file in /scripts/telemetry/inav/cfg.

JupaCreations, 2020-06-20:
HI, I performed your steps on the TX and confirm the script is working again.
Still unknown why it happens.

Thanks for the assistance and solution.

Paco Raap

teckel12, 2020-07-07:
Sounds like the configuration `model.dat` got corrupted.  I've never experienced this, but it can happen (SD cards are prone to corruption as well).  I would love to see the file.  If it happens again, please get me a copy of the dat file.

JupaCreations, 2020-07-08:
Hi, next time I will save the data from the SD card for examination and debugging.
But the removal of the model.dat is an easy but dirty solution for now.

BTW thanks again for a beautyfull script.

Paco Raap

JupaCreations, 2020-07-19:
Problem happened again. When switching from ARwing900 to CS Wing1000.
All model.dat files that were in the folder are ipped.
[lua script.zip](https://github.com/iNavFlight/LuaTelemetry/files/4943966/lua.script.zip)
Let me know if you can find something.

Thanks, Paco

patemil, 2020-08-29:
Got the same issue today, compared with another setup and my .dat file had a -1 value in last position that in the other configuration was 0. Changed it to 0 and it started to work again. 
Seems like the script can't handle negative values on one of the parameters.

@JupaCreations I see the same issue in your CSwing1000.dat

JupaCreations, 2020-09-02:
I currently run a Radiomaster TX16S with the script too. Sofar did had this problem.

Stringbrakr, 2020-12-08:
had the same problem today .

teckel12, 2021-02-16:
There's a PR for this #424

PR #424 (2021-02-16): Prevent negative config values from being saved
Fixes #389
commit a0bd3db3 (2021-02-16): Prevent negative config values from being saved

Fixes #389
closed 2021-02-16 by teckel12