teckel12/LuaTelemetry #390: Unable to playAudio() after update to OpenTX 2.3.8  with "Path too long" error message

paulo-sousa, 2020-06-14:
### Describe the bug
I just upgraded from OpenTX 2.3.7 to 2.3.8, copying the LuaTelementry-en files and configuration when upgrading the SD Card contents. I am using a Taranis Q X7 radio, with language in English.

### To reproduce
1. Upgrade a Taranis Q X7 to OpenTX 2.3.8.
2. Copy the iNav Telemetry Script 1.7.2 (LuaTelementry-en.zip) over to SD Card.
3. Configure a model with FrSky Telemetry.
4. Activate iNav Telemetry Script.
5. Turn model on, so that telemetry is received.
6. Perform a function that would trigger a voice alert (e.g. "Arm Motors").
7. **"Path too long"** error message is displayed on screen.

### Expected behavior
The sound alert should be heard and no error message should be shown.

### Screenshots
[image]

### Radio and model settings
[RadioAndTelemetrySettings.zip](https://github.com/iNavFlight/LuaTelemetry/files/4776728/RadioAndTelemetrySettings.zip)

### Additional context
OpenTX 2.3.8 was released yesterday. I reverted back to OpenTX 2.3.7 and the problem was gone (sound played correctly).

paulo-sousa, 2020-06-14:
Retested with OpenTX 2.3.9, which was released a few minutes ago and issue keeps occurring.
Also happens when selection other non-english voice languages.

paulo-sousa, 2020-06-14:
Retested with OpenTX 2.3.9, which was released a few minutes ago and issue keeps occurring.
Also happens when selecting other non-english voice languages.

paulo-sousa, 2020-06-14:
I printed out the call to playFile() and it seems correct, so the issue seems to be with OpenTX not accepting the around 33~38 characters.

In this example, it fails with this file:
**/SCRIPTS/TELEMETRY/iNav/en/good.wav**

[image]

paulo-sousa, 2020-06-15:
This seems to be an issue with OpenTX 2.3.8 & 2.3.9.

I opened an issue in the OpenTX project:
["Path too long" error when calling playFile() with 37+ characters ](https://github.com/opentx/opentx/issues/7739)

Stormdad, 2020-06-23:
I am also having the same issue after upgrading to OTX 2.3.9.

afterhender, 2020-06-24:
Hi, I have the exact same problem.
I'm using a Taranis Qx7s. Any solution found ?

paulo-sousa, 2020-06-24:
I submitted a bug directly to OpenTx with detailed information. This was immediately fixed and scheduled for [OpenTX 2.3.10](https://github.com/opentx/opentx/milestone/58?closed=1), but this milestone of OpenTX hasn't been released yet.

As soon **2.3.10** is released, you can update the firmware and the issue should be automatically fixed.

afterhender, 2020-06-24:
Great news, thank you

Stormdad, 2020-06-24:
> I submitted a bug directly to OpenTx with detailed information. This was immediately fixed and scheduled for [OpenTX 2.3.10](https://github.com/opentx/opentx/milestone/58?closed=1), but this [...]

Thank you for getting this issue address ! I use this script A LOT and look forward to it working again.

teckel12, 2020-07-07:
Nice job everyone identifying this and getting it fixed in OpenTX!  Due to this and a few other issues in the 2.3.x cycle, I'm going to require all future versions of LuaTelemetry run on OTX 2.3.10+

closed 2020-08-30 by stale[bot]