teckel12/LuaTelemetry #393: Tbs Tango 2

angelolorefice, 2020-07-03:
Hello on tango 2 it is not possible to enter the menu to configure thanks angel

xermalk, 2020-07-07:
> Hello on tango 2 it is not possible to enter the menu to configure thanks angel

The script doesn't have support for the tango 2. But i would love to get support. While its screen is physically small, the resolution is 128*96, more then the qx7.

xermalk, 2020-07-07:
[image]
Lots of unused space on the tango 2, but script loads up fine and you can change pages.

You just cant navigate up/down to adjust things.

teckel12, 2020-07-07:
The config menu doesn't work because OpenTX didn't include legacy controls.  The same is true with the Jumper T12.  There's a solution, but it breaks versions of OpenTX prior to 2.3.  Basically, this is a duplicate it #349

angelolorefice, 2020-07-07:
the main problem is the setting menus which makes the script unusable if you fly with lion batteries

teckel12, 2020-07-07:
@angelolorefice Does throttle stick up and to the right work?

The only other current solution is to use OpenTX 2.2 as OpenTX made the Jumper T12 not backwards compatible with 2.3+

There's plans to make it work, but it will cause anyone using OpenTX 2.2 for the script to fail with **ALL** transmitters, so I don't want to rush into a fix that corrects something for one transmitter with certain versions but breaks all other transmitters.

angelolorefice, 2020-07-07:
no it doesn't even work with gas and  right
thank you for your support

teckel12, 2020-07-07:
I guess I made that a Horus-only thing.  It's probably time to update to use the new 2.3+ control commands.

angelolorefice, 2020-07-08:
I have version 1.13 of freedom tx but I don't know which version of open tx it corresponds to

xermalk, 2020-07-11:
> I guess I made that a Horus-only thing. It's probably time to update to use the new 2.3+ control commands.

Does the script save all settings on the sd card? Aka could i set up the script on my qx7, and then move the folders to a tango 2 as a workaround? right now i'm getting quite a bit of warnings id like to disable/adjust for my li-ion powered plane :)

angelolorefice, 2020-07-13:
no don't save on sd i have already done this test

notamedclosed, 2020-07-22:
> Does the script save all settings on the sd card? Aka could i set up the script on my qx7, and then move the folders to a tango 2 as a workaround? right now i'm getting quite a bit of warnings id like to disable/adjust for my li-ion powered plane :)

For those like me that just want to change settings the config file for each model is saved on the SD card.
/SCRIPTS/TELEMETRY/iNav/cfg/[model name].dat

You have to have tried to open the config from the radio once for the file to be made.  (Menu button when script is running)

Open this with a text editor (your computer may try and use some other app so ensure you click open with..)

It's not easy to read, there are just a string of characters.  To decipher them you need to reference this file:
[https://github.com/iNavFlight/LuaTelemetry/blob/master/src/iNav/config.lua](https://github.com/iNavFlight/LuaTelemetry/blob/master/src/iNav/config.lua)
and this
[https://github.com/iNavFlight/LuaTelemetry/blob/master/src/iNav/config.lua](https://github.com/iNavFlight/LuaTelemetry/blob/master/src/iNav/menu.lua)

So the string goes like this (for example):
13534230400010121100020301025003100150004300000

Each line of local config in config.lua tells us how to read it.  So for example the first line 
{ o = 1,  c = 1, v = 1 }, -- Battery View - 1

Tells us that the battery view variable is 1 character (C=1) and the default is 1 (V=1).  
Under the menu.lua you can see the various options: 0 = cell, 1 = total battery.

The next line tells us to read 2 characters:
{ o = 3,  c = 2, v = 3.5, d = true, x = 3.9}, -- Cell Low - 2

Tells us at what value the cell value is considered low, you can see that mine is 35 (and d=true means this is a decimal) so the value is 3.5.

Keep reading through the string and adjusting settings.
So if my string was this at the start
135341....

Then I have battery view set to 1, 3.5v as low, 3.4 as critical, and I have option 1 set for Voice Alerts which means critical only.

It's not super intuitive but you can then change things so you don't have to listen to alarms blaring during flight.

closed 2020-09-12 by stale[bot]

sweebee, 2021-05-21:
Stil isn't fixed right? Cant use the menu on my Tango 2.

hcolsen, 2023-01-04:
Anything new on The sub?