teckel12/LuaTelemetry #397: FM value from CRSF telemetry not in log file.  Log playback doesn't work

tonyyng, 2020-07-26:
### Describe your question
My hardware combination is a Jumper T16 with Crossfire Micro TX V2 and Crossfire Nano receiver.  Looking at the source code in crsf.lua, it interprets the FM field as a set of strings (OK, ACRO, WP, HOLD, etc).  I see these values when I add the FM field as a widget.

However, I don't see any values like that stored in the log file.  All the values in my log file have FM value "0".  Here are a few lines:

```
Date,Time,Ptch(rad),Roll(rad),Yaw(rad),RxBt(V),Curr(A),Capa(mAh),Bat_(%),1RSS(dB),2RSS(dB),RQly(%),RSNR(dB),RFMD,TPWR(mW),TRSS(dB),TQly(%),TSNR(dB),FM,GPS,GSpd(kmh),Hdg(@),Alt(m),Sats,VSpd(m/s),Rud,Ele,Thr,Ail,S1,6P,S2,EX1,EX2,LS,RS,SA,SB,SC,SD,SE,SF,SG,SH,LSW,TxBat(V)
2020-07-25,18:16:48.870,-0.30,-0.08,-0.71,7.8,0.0,0,77,68,68,94,25,1,10,49,100,30,0,43.486963 -80.604234,26.1,-31.43,71,19,0.6,-2,0,-213,-2,-23,-1024,-34,0,0,442,1024,-1,0,0,0,-1,-1,-1,-1,0x0000000000000000,7.7
2020-07-25,18:16:49.420,-0.30,-0.08,-0.71,7.8,0.0,0,77,67,67,94,18,1,10,47,100,21,0,43.487036 -80.604281,27.8,-32.49,75,19,0.6,-2,0,-213,-2,-23,-1024,-34,0,0,442,1024,-1,0,0,0,-1,-1,-1,-1,0x0000000000000000,7.7
```
Because of this, the playback of the log doesn't do the same things.

I think I'm on the current versions of the various firmwares (Open Tx 2.3.9, CSRF V 3.74 and 1,7,2 of LuaTelemetry)

Have I misconfigured something?  Can you point me in the right direction?

Thanks

tonyyng, 2020-07-26:
It looks like you've previously found this problem: https://github.com/opentx/opentx/issues/6413

It might be worth a mention on the log playback documentation until the problem is fixed.

teckel12, 2020-07-31:
The problem is that OpenTX doesn't save the FM field in the log files because it's a text string.  It's obviously a low priority to the OpenTX team and they refuse to merge changes from outsiders so I'm not going to waste my time with it.

closed 2020-07-31 by teckel12