teckel12/LuaTelemetry #409: Activating iNav widget changes text color on other widgets

RC-SOAR, 2020-08-20:
### Describe the bug
Activating iNav widget changes text color on other widgets. 
Radio: Jumper T16, Horus X12S

### To reproduce
1. Install iNav on SD card
2. Go to Telemetry menu
2. Select default OpenTX theme, set background to white, main color to any dark color
3. Configure page 1 for channel display widget
4. Configure page 2 for iNav widget
5. Press {Rtn} to exit to page 2
6. Power cycle radio. On restart, displays page 2
7. Press {Page up} to show telemetry page 1
8. Press {Page up} to show telemetry page 2

### Expected behavior
Page 2 text color was black, and should remain black. But it changes to white.

The same occurs with ShowItAll widget (as supplied with  TX16S)

### Screenshots
Screenshot 1: default theme properties
Screenshot 2: channel widget before
Screenshot 3: page to iNav
Screenshot 4: page back to channel widget

[image]
[image]
[image]
[image]

Hobby4life, 2020-10-04:
Can confirm this

xxJohnxx, 2021-02-12:
Can confirm, this happens on the RadioMaster TX16S as well.

porlock, 2021-02-15:
i have it to on RadioMaster TX16S

expipiplus1, 2021-02-16:
Should this be considered an opentx bug, improper sandboxing of widgets.

wizardrnd, 2021-02-16:
See #426 for a mitigation - enabling the "restore" setting under the widget settings.

closed 2021-02-16 by teckel12