teckel12/LuaTelemetry #410: Possible Timing Problem

RomanFoltyn, 2020-08-28:
**Hallo,**
Tx model Jumper T18 TBS Crossfire  test   with diverse Matek F765 and F35 Lighting 
Battery 3S 12,7 V and Labor Powersupply 10V
To be sure, all servo and motor are disconnected. TBS Crossfire Firmware 4.0 
RX -> TBS Micro V2 and TBS Nano Diversity

With all these combinations, the problem was exactly the same.

Problem with measuring battery voltage
every few seconds (irregularly) the voltage telemetry display jumps to the value of about 4.2 V which causes that every moment a message about bad battery condition is reproduced. 
The spikes in tension are not real. 
there is probably a timing problem. 
Other indications by telemetri show no irregularities and shown without spikes. 
Battery status in % is shown without spikes.

Please Help,
Roman Foltyn (DL6NBF)

RomanFoltyn, 2020-08-30:
Hallo
**For me to use my system it would be enough to turn off the airplane battery alarm**

RomanFoltyn, 2020-09-02:
It doesn't work with TARANIS 9d PLUS either - always the same problem

wx4cb, 2020-09-28:
make sure you're on the dev version

teckel12, 2021-02-16:
Duplicate #411

closed 2021-02-16 by teckel12