teckel12/LuaTelemetry #411: Probable problem with reading CSRF telemetry by Jumper T18

RomanFoltyn, 2020-08-30:
Hallo,
Tx model Jumper T18 TBS Crossfire test with diverse Matek F765 and F35 Lighting
Battery 3S 12,7 V and Labor Powersupply 10V
To be sure, all servo and motor are disconnected. TBS Crossfire Firmware 4.0
RX -> TBS Micro V2 and TBS Nano Diversity

With all these combinations, the problem was exactly the same.

Problem with measuring battery voltage
every few seconds (irregularly) the voltage telemetry display jumps to the value of 4.3 V which causes that every moment a message about bad battery condition is reproduced.
The spikes in tension are not real.
there is probably a timing problem or CRSF telemetry Decoding by T18
Other indications by telemetry show no irregularities and shown without spikes.
Battery status in % is shown without spikes. OLED Display (Matek F765) is stable.

For me to use my system it would be enough to turn off the airplane battery alarm
The alarm going off all the time (Vibrate and Mesage) is unbearable.

Please Help,
Roman Foltyn (DL6NBF)

teckel12, 2021-02-16:
@RomanFoltyn It should just be reporting what it's receiving.  It's possible that it's not stable but not visible that it's changing.

You can turn off alerts for specifically the battery or other settings in the config menu:

https://github.com/iNavFlight/LuaTelemetry/wiki/Configuration-Settings

wx4cb, 2021-02-21:
turn of RxBatt sensor in the crossfire menu - i beleive it's in the receiver. but then the script doesn't do that. it's a lovely tbs oddity

closed 2021-04-14 by stale[bot]