teckel12/LuaTelemetry #416: Radiomaster TX16S Model Names disappear in model screen after Lua install

terrorswellfpv, 2020-09-25:
Using latest version 1.72

Radiomaster Open tx version is 2.3.8

[image]

JupaCreations, 2020-09-25:
Same problem. If it is LUA related no idea but model names sometimes disappear.

mastersurferde, 2020-09-25:
Hi,
I have the same problem with the Jumper T16 and the actual OpenTX 2.3.10N3xx (also with previous nightly versions and official versions). The proplem is always present if you use a model with crossfire (and lua-telemetry) and you disconnect the crossfire module bevore you change the model.
Best regards,

Gaylian, 2020-09-30:
I have same issue with Jumper T16. See issue #404. I placed an issue with OpenTx and their reply was it's a telemetry LUA problem. They have seen this issue before.!

ojoojoo, 2020-10-20:
Same problem on a Radiomaster TX16S, model names dissapear

terrorswellfpv, 2020-10-20:
Change the color of the background to blue and model names will reappear. This is the fix until it gets patched.

ojoojoo, 2020-10-20:
Thanks for the answer, selecting a default widget and rebooting the radio also works.

Tygrys-1, 2020-10-27:
It is not the model names disapear, but the font color is changed to white.

FrankPetrilli, 2021-02-02:
See #426 for a mitigation - enabling the "restore" setting under the widget settings.

teckel12, 2021-02-16:
Correct, this is an issue with OpenTX as the color settings are global.

The solution is https://github.com/iNavFlight/LuaTelemetry/wiki/Installation#installsetup-lua-telemetry-on-transmitter under `Install/Setup Lua Telemetry on Transmitter` > `Horus/Jumper T16` > `g`:

Optionally (while still selecting the iNAV script), long-press Enter, select Widget settings where you can enable Restore (to restore your theme's colors) and set your theme's Text color and Warning color

closed 2021-02-16 by teckel12