teckel12/LuaTelemetry #419: Spurious battery warnings when fuel sensor is removed

skorokithakis, 2020-10-28:
I have a RadioMaster with R9 and an F405-WING FC. I removed the fuel sensor, but whenever I arm I get "battery critical, zero percent", which shows me that Lua Telemetry is still trying to read from the fuel sensor. Adding the sensor back with the units on the FC set to `PERCENT` fixes the issue.

skorokithakis, 2020-12-25:
No

teckel12, 2021-02-16:
The fuel sensor is required.  It's not designed to work without a fuel sensor.

skorokithakis, 2021-02-16:
Hmm, I was going by [your comment here](https://github.com/iNavFlight/LuaTelemetry/issues/353#issuecomment-542025852), is that not the case then?

teckel12, 2021-02-16:
Are you rebooting your transmitter after removing the sensor?

skorokithakis, 2021-02-16:
I did, but I'll try again and let you know.

closed 2021-04-09 by stale[bot]