teckel12/LuaTelemetry #422: Support PlusCodes Coordinate display

danarrib, 2020-11-14:
INAV Lua Telemetry is super nice. But it would be awesome if it was able to display GPS Coordinates using Plus Code notation.

Plus Code (a.k.a. Open Location Code) is a coordinate notation that converts standard GPS coordinates into easily readable text. 

It's already supported by INAV OSD (https://github.com/iNavFlight/inav/pull/4158). 
And of course, it's supported by Google Maps.

One of the biggest reasons to use it is that it's easier to copy from Taranis Screen a Plus code than a pair of numbers from latitude and longitude. It would help to recover a crashed aircraft faster if operator doesn't have to type the complicated numbers from GPS coordinates.

Example:
Aircraft crashed at Coordinates **-23.470937,-46.832563**
Instead of typing this number on Google Maps, I can just type it's Plus Code equivalent, which is **588MG5H8+JX**, which has only 11 chars.

It also saves space on the screen by making the coordinates a single-line element.

You can check it in deep by visiting https://github.com/google/open-location-code 
There are libraries for lots of different languages, but I'm not sure if there's a LUA library tho... Maybe it would need some extra work to port from another language.

teckel12, 2021-02-16:
I like the idea, but will need to look at any challenge to implement.  The encoding is one thing, but the limited space is another issue (uses two lines for latitude/longitude while this would be a single line, and probably too long to display).

teckel12, 2021-02-16:
I just looked at the code required and it seems quite extensive for the poor little microcontroller on the transmitter.  As LuaTelemetry is already pushing it's abilities, I fear adding this would overload the little guy.  The code would also need to be refactored from the currently supported languages to Lua, which would require some effort.  I'd be willing to see how it could be implemented in Lua however.

closed 2021-04-10 by stale[bot]