teckel12/LuaTelemetry #425: scroll config Menu X9D+ 2019

emics, 2020-11-19:
When i try to scroll the menu the cursor doesn't move

### Describe the bug
From Initial Screen I long press PAGE and I see the Classic View. After I press MENU button but if i scroll the jog the cursor doesn't move. If i press the jog I can see the first entry "Battery View" TOTAL blinking.

LUA Version 1.7.2
Taranix X9D+ 2019 opentx  2.3.10

my .cfg file contain this:
`13534230120011121110020302035003000150004300000`

RotorVulura, 2021-01-09:
Hi !
Same here. Taranis X9D+ 2019 openTX 2.3.10
Also, I can not switch between the views pressing the jog. It just stuck at first type of view.
I can reset telememetry or flight properly by holding jog button and then scroll to the function desidered {and press jog again to confirm}.
Development build seems to works. Not tested in flight yet.

teckel12, 2021-02-16:
Duplication of #349

closed 2021-02-16 by teckel12