teckel12/LuaTelemetry #426: Tx16s lua makes model names etc white (unreadable)

erikjede1ste, 2020-12-12:
### Describe your question

### To reproduce

1. Go to '....'
2. Click on '....'
3. Scroll down to '....'
4. Should it be doing this?

### Expected behavior

### Screenshots

### Additional context

b14ckyy, 2021-01-20:
Same issue here. And also on a Horus X10S with OpenTX 2.3.9. I opened a OpenTX ticket first as the Horus user was mistaken and said it also happens without the LUA. But he can reproduce the issue in the same way. Here my full Issue description: 

Homescreen font color switches from black to white randomly. Can be triggered with LUA Widget.
This applies for multiple color screen radios

To Reproduce
Steps to reproduce the behavior manually:

1. Set up a new model
2. keep the first home screen page as it is (with black fonts and table frames)
3. add a second screen with the INAV LUA Telemetry Widget (no rc link or any setup needed)
4. switch back to first screen (issue might be visible already)
5. turn off radio and turn it back on
6. Fonts on first home screen are normal black now as they should be
7. switch to the second screen to show the INAV LUA widget
8. switch back to first screen
9. font color, table frames, sliders and trim position bars are all white now
10. A Reboot temporarily solves issue.

If a bright background image is used, the font won't be visible anymore.
Also the system menu and model setup menu font can switch to white and then the menu entries are not visible anymore till they are hightlighted.
This also happens with different themes so it is not Theme related.

Expected behavior
Font should stay in the same color all the time.

Screenshots
Will add screenshots later if needed

General information

OpenTX 2.3.11 on Radiomaster TX16S
OpenTX 2.3.9 on Horus X10S
OpenTX 2.3.8 (Factory) on TX16S (need confirmation from a friend who has not updated yet)

b14ckyy, 2021-01-26:
just a gentle question to @teckel12 
Do you thik you will find time to fix this issue? Otherwise I might ask a friend of mine who also can program LUA to have a look and help.

wizardrnd, 2021-01-26:
Try to configure wigget ;)
`Optionally (while still selecting the iNAV script), long-press Enter, select Widget settings where you can enable Restore (to restore your theme's colors) and set your theme's Text color and Warning color`
https://github.com/iNavFlight/LuaTelemetry/wiki/Installation#installsetup-lua-telemetry-on-transmitter

b14ckyy, 2021-01-26:
@wizardrnd OMG You are a hero! 
Sometimes it might be helpful to RTFM. Damn I was wondering why there is no widget settings menu by a single click like the others. The restore option fixed. Thanks so much!

erikjede1ste, 2021-01-26:
@wizardrnd . Yeah that fixed it man!  Thank you very much!

b14ckyy, 2021-01-26:
> @wizardrnd . Yeah that fixed it man! Thank you very much!

I'd suggest to close this issue now as it is resolved.

closed 2021-01-26 by erikjede1ste

teckel12, 2021-02-16:
Correct, this is an issue with OpenTX as the color settings are global.

The solution is https://github.com/iNavFlight/LuaTelemetry/wiki/Installation#installsetup-lua-telemetry-on-transmitter under `Install/Setup Lua Telemetry on Transmitter` > `Horus/Jumper T16` > `g`:

Optionally (while still selecting the iNAV script), long-press Enter, select Widget settings where you can enable Restore (to restore your theme's colors) and set your theme's Text color and Warning color