teckel12/LuaTelemetry #427: Not Ok Message after Update to Open TX 2.3 and Inav 2.6

SteffenFPV, 2020-12-15:
After Update to Open TX 2.3 and Inav 2.6 I get the "NOT OK" message on all screens.

Parallel check with Inav Configurator showes that all pre-arming checks were passed, hardware is healty and navigation is safe (enough sattelites).

Also it is possible to arm motors and start flying.

But the "NOT OK" Message stays on all telemetry screens, also during flight.

SteffenFPV, 2020-12-31:
I found a solution in source code.

In File Inav.lua replace line 229:

data.modeId = (bit32.band(modeE, 2) == 2 or modeE == 0) and (data.throttle > -920 and 12 or 5) or 6 -- Not OK to arm(5) / Throttle warning(12) / Ready to fly(6)

by

if (data.throttle > -920) then
    data.modeId = 12 --Throttle warning(12)
elseif (bit32.band(modeE, 2) == 2 or data.gpsFix == false) then
    data.modeId = 5 --Not OK(5)
else	
    data.modeId = 6 --Ready to fly(6)
end

michaelj0hn, 2021-01-25:
@SteffenFPV There was an issue in OpenTX and Flightmode Sensor. I had experienced the same issue as you described, the "FM" sensor value was not processed correctly. This got fixed in OpenTX 2.3.11 now.
I am using the LuaTelemerty with Betaflight and it started working again.
This might be related to it: https://github.com/opentx/opentx/issues/8170

closed 2021-04-18 by stale[bot]