teckel12/LuaTelemetry #428: Not able to set Battery Capacity on Taranis Q X7S

pDR8iLQ4go, 2020-12-23:
I tried to set the battery capacity as explained here: https://github.com/iNavFlight/LuaTelemetry/wiki/Configuration-Settings

But I am not able to enter any value (e. g. 5500mAh) using my Taranis Q X7S. I've seen the SC card folder E:\SCRIPTS\TELEMETRY\iNav\cfg and expected that I'm able to change some parameters by using a text editor. But the configuration files (*.dat) do not have a human readable structure. Is there a way to change the default value (1500mAh) by editing any text file?

pDR8iLQ4go, 2020-12-23:
Am I allowed to change the battery capacity in this config file in order to change the value?
[image]

Mateyhv, 2020-12-27:
The same happens on the Taranis X9D so that must be a bug with the black and white screen radios. There is no way to set the Battery Capacity parameter in the setup menu.

pDR8iLQ4go, 2020-12-29:
@Mateyhv: do you know if it is possible to edit the configuration file (in my case „Tarot.dat“)?

Mateyhv, 2020-12-31:
@pDR8iLQ4go I am not sure. The best outcome is the bug to be corrected.

teckel12, 2021-02-16:
This has to do with breaking changes with OpenTX firmware having to do with the transmitter buttons.

teckel12, 2021-02-16:
Duplication of #349

closed 2021-02-16 by teckel12