teckel12/LuaTelemetry #433: HORUS X10 & X12 INAV Lua Disables other Lua Display Scripts

jimbo385inav, 2021-01-26:
### Describe the bug

### To reproduce

1.  Create and only have 2 models on the Horus. 
2.  Load INAV Lua on 1 model. 
3. Displays fine on this model
3. Add second model with a screen just containing the Top Bar and/or Sliders & Trims
4. Display has Top Bar present and the Sliders & Trims only
5. Where Widgets can be displayed is "Disabled"
6. Widgets can be added to second screen but only  "Disabled" is on the screen

### Expected behavior
 INAV Lua should not affect other model screens on the TX. One should be allowed to have both UNAV and other Telemetry screens for other models and if required, the same model.
INAV is expecting that all model Telemetry screens on the HORUS must be full screen and not be able to display the Top Bar and/or Sliders & Trims.  

### Screenshots
I have a couple of videos that I've loaded to FB. 

### Radio and model settings
 Horus X10 Express, ISRM 2.x. Version 2.3.10. Latest INAV Lua 

### Additional context
 I posted this to Facebook which contains additional information and videos of the issue.
FB Page is "FrSky Horus X10 / X12 Users Group" and the issue is documented 22 January 2021 at 00:59

teckel12, 2021-02-16:
You can run different widgets.  LuaTelemetry doesn't change the views of other widgets.

jimbo385inav, 2021-02-16:
Hi Tim,

Thanks for your reply but I'm afraid that you are incorrect!

If I have 1 model using the INAV Lua Script, any other model with a
Telemetry screen that includes either the Top bar and/or the Sliders and
Trims, these screens will not display. "Disabled" in red will be displayed
instead of the screen! I've tried it on another Horus with the same result!

It appears that the INAV Lua is performing checks even on another mode that
does not have the INAV script.

Regards,

Jim
Please visit
Oddjobhandymanservices.co.uk

teckel12, 2021-02-16:
@jimbo385inav I'm not experiencing that.  I use multiple widgets for multiple models and each widget works correctly, including LuaTelemetry.  For the page that LuaTelemetry is on, that needs to be full screen (no top bar or trims).  But other widget pages can be different settings.  LuaTelemetry isn't doing something to disable other widgets or screens, and model to model can be different.

Maybe a screenshot as maybe I'm not understanding you correctly.

jimbo385inav, 2021-02-16:
Hi Tim,

I've done some more testing to ascertain when things fail!Firstly, it
appears that the INAV script works fine. The culprit appears to be another
script called Showitall. If I have the INAV script running and have the
Showitall script on another screen and on another model. It works fine. If
I select the other model with Showitall on it, that also works fine. If I
switch off and switch back on again, the Showit all screen is disabled on
both models! I've done loads of permutations and that's what keeps on
happening!

So, for now, I need to chat with the creator of the Showitall script to try
and get to the bottom of this.

Again, many thanks for your responses.

Regards,

Jim
Please visit
Oddjobhandymanservices.co.uk

teckel12, 2021-02-16:
@jimbo385inav Glad this seems to be sorted out.  I'll check out the Showitall script and see if I can find out what's going on as well.  I'll close this issue as well as https://github.com/teckel12/LuaTelemetry/issues/20 as it seems it's not a LuaTelemetry issue.

closed 2021-02-16 by teckel12

jimbo385inav, 2021-02-16:
> what's going on as well. I'll close this issue as well as teckel12#20

That would be great Tim if you can check out the Showitall script as I
would like both on my Horus! Lol

Here is the Showitall script.

https://rc-soar.com/opentx/lua/showitall/index.htm

Regards,

Jim
Please visit
Oddjobhandymanservices.co.uk