teckel12/LuaTelemetry #437: Make an option to choose interface language in settings of Lua Telemetry

nm17, 2021-02-16:
### Is your feature request related to a problem? Please describe

Currently, OpenTX has no Russian language and it has no way of selecting a language that's not implemented yet, so my interface translations to Russian are useless, since Russian cannot be selected as a OpenTX menu language.

### Describe the solution you'd like

A good solution to this problem would be to have the OpenTX language be the default language in Lua Telemetry, but also have an option to choose which language to use for the interface in Lua telemetry settings.

### Describe alternatives you've considered

N/A

### Additional context

My pull request to add Russian language: #405

teckel12, 2021-02-16:
I show Russian as one of the languages in OpenTX.

https://github.com/opentx/opentx/blob/a07fb2bb5abf5db24007fcbf875a1a35156ce545/companion/src/generaledit/generalsetup.cpp#L339

nm17, 2021-02-16:
> I show Russian as one of the languages in OpenTX.
> 
> https://github.com/opentx/opentx/blob/a07fb2bb5abf5db24007fcbf875a1a35156ce545/companion/src/generaledit/generalsetup.cpp#L339

Thats Voice languages, not menu languages

teckel12, 2021-02-16:
@nm17 May I suggest that Russian menu language support is added to OpenTX then?

https://github.com/opentx/opentx/tree/2.3/radio/src/translations

nm17, 2021-02-16:
> @nm17 May I suggest that Russian menu language support is added to OpenTX then?
> 
> https://github.com/opentx/opentx/tree/2.3/radio/src/translations

You can, I can translate it myself even, but in my opinion most Russians are already used to the English interface. I think the simpler option that doesn't require much work is to add an option to chose the language in lua telemetry itself.

teckel12, 2021-02-16:
@nm17 LuaTelemetry is hitting a resource wall and any additions are risky as other things may fail.  Are you even sure the Russian language font works?  When I've tried using special characters, it didn't work.  In other words, if you force Russian in LuaTelemetry (via changing `data.lang = "ru"` here -> https://github.com/iNavFlight/LuaTelemetry/blob/6da9fca810f5a5837722bc1d5a230b8611b9536d/src/iNav.lua#L36) does it even display the Russian special characters correctly?  Maybe it doesn't work for me because I'm using the companion.

teckel12, 2021-02-18:
@nm17 Support for this has been added in the latest development branch:

https://github.com/teckel12/LuaTelemetry/wiki/Upgrade-to-Development-Build

nm17, 2021-02-19:
Will merge later

closed 2021-02-19 by nm17