teckel12/LuaTelemetry #439: Why does LuaTelemetry widget not work on OpenTX firmware simulator?

wizardrnd, 2021-02-18:
I tried running it by simulate opentx-tx16s, opentx-t16, opentx-x10 and opentx-x12s with OpenTX 2.3.10 and 2.3.11, but it doesn't show up on selecting page widget

teckel12, 2021-02-18:
If LuaTelemetry isn't listed as a Widget, it was installed in the incorrect location. Please see the following instructions on the exact steps to install to your transmitter:

https://github.com/teckel12/LuaTelemetry/wiki/Installation#installsetup-lua-telemetry-on-transmitter

wizardrnd, 2021-02-18:
LuaTelemetry is listing and working correct on my "real" Radiomaster TX16S, but it not listing and not working on OpenTX firmware simulator. Another LUA scripts and widgets (for example, FrskyTelemetryScript and HorusMappingWidget by yappu) are listing and working on simulator too. But iNAV LuaTelemetry is working on real device only.

teckel12, 2021-02-18:
https://github.com/teckel12/LuaTelemetry/issues/22#issuecomment-781359515

nm17, 2021-02-19:
Can confirm that I have the same issue on the simulator

teckel12, 2021-02-19:
I do not experience this, it's my primary development environment.  Keep in mind, LuaTelemetry is currently designed for OpenTX 2.2 due to breaking changes in OpenTX 2.3.

closed 2021-04-19 by stale[bot]