teckel12/LuaTelemetry #440: Jumper T12

kasatka60, 2021-02-19:
Jumper T12 pro openTX 2.3.11
LuaTelemetry v1.7.3
How to enter the menu?

nm17, 2021-02-19:
@kasatka60 have you tried going into the menu like it is described in the docs?

https://luatelemetry.readthedocs.io/en/latest/Configuration-Settings/#horusjumper-t16

teckel12, 2021-02-19:
@kasatka60 Not sure about this fork, but my v1.7.4 build was supposed to resolve this, you can [get it here](https://github.com/teckel12/LuaTelemetry).

kasatka60, 2021-02-20:
Haven't tried it yet. Have you implemented menu entry as a jumper-t16? 
I will try.

nm17, 2021-02-20:
> Haven't tried it yet. Have you implemented menu entry as a jumper-t16?
> I will try.

It should theoretically work on any transmitter I think, so give it a try.

PasiRuo, 2021-02-20:
works ok here with  [https://github.com/iNavFlight/LuaTelemetry/pull/400](url)

kcneato, 2021-02-20:
I have similar issue on Jumper t lite. Cant access the menu button with sys neither the stick comands.

kasatka60, 2021-02-20:
> It should theoretically work on any transmitter I think, so give it a try.

Don't work

kasatka60, 2021-02-20:
> I have similar issue on Jumper t lite. Cant access the menu button with sys neither the stick comands.

Try this:
https://yadi.sk/d/yMzA1EA6blM2EQ

kcneato, 2021-02-20:
> > It should theoretically work on any transmitter I think, so give it a try.
> 
> Don't work

i tried the stick commands with the tarranis to get the menu up but did not work (the button works great). Maybe something  wrong with t lite opentx image still, waiting for the official updated for the jumper

nm17, 2021-02-20:
We are working on some new important stuff for the repo, so expect some delays with this. I'll make sure this gets done ASAP.

closed 2021-04-19 by stale[bot]