teckel12/LuaTelemetry #442: EMUFLIGHT support

Xedos9er, 2021-02-26:
I tried the script with emuflight and noticed that it does not work.
Looks like there´s no indication of arm status transfered via flight mode channel. 
Betaflight uses "*" to indicate arm status.

Any plans to support emuflight in future or whats needed to support it?

teckel12, 2021-02-27:
@Xedos9er First to clarify, your request pertains only to Crossfire telemetry, not FrSky.

Secondly, it's not so much that LuaTelemetry needs to support EmuFlight, it's that EmuFlight needs to be updated to output telemetry information like INAV and Betaflight, at which time, LuaTelemetry will automatically work with EmuFlight.

Looking at the EmuFlight code, it seems it was forked before I submitted the pull request to Betaflight to enhance the flight modes.  At first glance, it appears the only change that would be required is the `crsfFrameFlightMode` function in the `src/main/telemetry/crsf.c` file.

The function in EmuFlight:
```
void crsfFrameFlightMode(sbuf_t *dst) {
    // write zero for frame length, since we don't know it yet
    uint8_t *lengthPtr = sbufPtr(dst);
    sbufWriteU8(dst, 0);
    sbufWriteU8(dst, CRSF_FRAMETYPE_FLIGHT_MODE);
    // use same logic as OSD, so telemetry displays same flight text as OSD
    const char *flightMode = "ACRO";
    if (isAirmodeActive()) {
        flightMode = "AIR";
    }
    if (FLIGHT_MODE(FAILSAFE_MODE)) {
        flightMode = "!FS";
    } else if (FLIGHT_MODE(ANGLE_MODE)) {
        flightMode = "STAB";
    } else if (FLIGHT_MODE(HORIZON_MODE)) {
        flightMode = "HOR";
    }
    sbufWriteString(dst, flightMode);
    sbufWriteU8(dst, '\0');     // zero-terminate string
    // write in the frame length
    *lengthPtr = sbufPtr(dst) - lengthPtr;
}
```

The function in BetaFlight:
```
void crsfFrameFlightMode(sbuf_t *dst) {
    // write zero for frame length, since we don't know it yet
    uint8_t *lengthPtr = sbufPtr(dst);
    sbufWriteU8(dst, 0);
    sbufWriteU8(dst, CRSF_FRAMETYPE_FLIGHT_MODE);

    // Acro is the default mode
    const char *flightMode = "ACRO";

    // Modes that are only relevant when disarmed
    if (!ARMING_FLAG(ARMED) && isArmingDisabled()) {
        flightMode = "!ERR";
    } else

#if defined(USE_GPS)
    if (!ARMING_FLAG(ARMED) && featureIsEnabled(FEATURE_GPS) && (!STATE(GPS_FIX) || !STATE(GPS_FIX_HOME))) {
        flightMode = "WAIT"; // Waiting for GPS lock
    } else
#endif

    // Flight modes in decreasing order of importance
    if (FLIGHT_MODE(FAILSAFE_MODE)) {
        flightMode = "!FS!";
    } else if (FLIGHT_MODE(GPS_RESCUE_MODE)) {
        flightMode = "RTH";
    } else if (FLIGHT_MODE(PASSTHRU_MODE)) {
        flightMode = "MANU";
    } else if (FLIGHT_MODE(ANGLE_MODE)) {
        flightMode = "STAB";
    } else if (FLIGHT_MODE(HORIZON_MODE)) {
        flightMode = "HOR";
    } else if (airmodeIsEnabled()) {
        flightMode = "AIR";
    }

    sbufWriteString(dst, flightMode);
    if (!ARMING_FLAG(ARMED)) {
        sbufWriteU8(dst, '*');
    }
    sbufWriteU8(dst, '\0');     // zero-terminate string
    // write in the frame length
    *lengthPtr = sbufPtr(dst) - lengthPtr;
}
```

I found the pull request where these changes were made, which should be a fairly simple pull request to submit to EmuFlight:

https://github.com/betaflight/betaflight/commit/4d8bf61d94614f20ff8ace847b5d2b12d8710bdf#diff-5b4ded67d6b41332a7188e3100bec940e517ea32550f01e25709b13f86eabc77

With that said, I don't use EmuFlight, and I'm not 100% sure that all the changes are available in EmuFlight.  They appear to be, but I can't compile and run EmuFlight to test.

I've [submitted a pull request to EmuFlight](https://github.com/emuflight/EmuFlight/pull/506) which should allow LuaTelemetry compatibility.

Xedos9er, 2021-02-27:
Thank you for figuring out what needs to be changed.

teckel12, 2021-02-28:
@Xedos9er Seems the PR works.  https://github.com/emuflight/EmuFlight/issues/481

Let me know if there's any issues.  I'll leave this open till the PR is merged.

closed 2021-04-21 by stale[bot]