teckel12/LuaTelemetry #453: support for expressLRS

nick9one1, 2021-03-14:
It could be great if we could use this with ExpressLRS and iNav. 

https://github.com/ExpressLRS/ExpressLRS/

teckel12, 2021-03-19:
@nick9one1 Is ExpressLRS supported in INAV?  It doesn't appear to be.  It would first need to be supported in INAV along with telemetry for it to have any chance of compatibility with LuaTelemetry.

nick9one1, 2021-03-20:
actually it looks like this is already supported :) 

this is with ExpressLRS

the telemetry screen on the TX shows data too. 

[image]

teckel12, 2021-03-22:
I don't see anything about ExpressLRS in INAV, nor do I have anything running ExpressLRS.  If it's showing telemetry, it would already be working if the telemetry matched that of FrSky.  If not, you'll need to create a pull request with the specifics for this telemetry.  Basically, you're not giving enough information to do anything and as I'm not flying ExpressLRS, there's no possible way I could do it.  So, someone flying ExpressLRS would need to verify all the telemetry which is being reported as a bare minimum, if not create a pull request which supports ExpressLRS (if it's even possible, which so far I have not been shown that it is).

sweebee, 2021-05-13:
it just works since it uses the crossfire protocol. Just make sure you've set the telemetry interval high enough.

Xedos9er, 2021-05-17:
Yes it works, but there needs to be done some minor code adjustments. Crossfire has only 2 modes (1=50 and 2=150hz), while ExpressLRS supports more than 2 modes:
1=25hz
2=50hz
4=150hz
6=250hz
7=500hz
8=1000hz ?? (not sure since it´s experimental)

Maybe @teckel12 adds a menu option to switch between CF and Express LRS or he manages somehow to autodetect ELRS ??
I also don´t know why mode 3 is missing?!

sweebee, 2021-05-17:
i'm not a LUA expert but it looks kinda basic:

```
text(LCD_W, 0, (data.rfmd == 2 and 150 or (data.telem and 50 or "--")) .. "Hz", RIGHT + tmp)
```

The lua script doesn't work very well on my Tango 2 so hard to test something, but I get a TX16s in the mail this week so i might try to convert it for ELRS. but again I'm not a LUA expert :P

teckel12, 2021-05-25:
Looks like Crossfire reports the uplink rate (rfmd) as the following:

0 = 4Hz, 1 = 50Hz, 2 = 150Hz

This isn't compatible with the ExpressLRS uplink rates.  If the value is 4 or higher, it could be assumed it's ExpressLRS.  But the inconsistency of the 1 & 2 values don't fit well.

closed 2021-07-16 by stale[bot]