teckel12/LuaTelemetry #460: Fuel reads wrong until widget restarted

iceboundflame, 2021-04-22:
### Describe the bug
My nearly-full 6s 1300mah battery often reads 0% fuel or 31% fuel when starting up.  The per-cell voltage is also wrong, for example at one point I was seeing 3.56 V per cell.  This leads me to believe maybe the cell count detection is wrong. I expected it to be a full or nearly full 6s battery.

I was able to resolve it by switching to another model in the OpenTX model selector and switching back. Then it started reading full fuel as expected.

Is there any documentation on how the battery settings are meant to be set? Thanks!

- RadioMaster TX16S, Micro Crossfire V2 module with Crossfire Nano RX 
- Cell Low 3.5V
- Cell Critical 3.4V
- Cell Calculation 4.2V (I think problem was happening when this was at 4.3V too)

[Crsf-RRiotHD1.dat.txt](https://github.com/teckel12/LuaTelemetry/files/6355933/Crsf-RRiotHD1.dat.txt)

teckel12, 2021-05-25:
If switching models works, it probably just means you need to do a reset between flights.

https://github.com/teckel12/LuaTelemetry/wiki/Tips-&-Common-Problems#tips

> * Between flights (before armed), long-press Enter/dial and select `Reset telemetry` to reset telemetry values

iceboundflame, 2021-05-26:
Thanks for the reply, I will try that next time I notice this. I found this sometimes happened on my first flight of a session though. Perhaps it has to do with the order in which I power up my transmitter vs the model.

closed 2021-07-17 by stale[bot]