teckel12/LuaTelemetry #461: lnva Telemetry script v1.7.4 shows “NOT OK” and no fields will populate

hprompt, 2021-05-06:
Hi there,
I know this has been asked several times before, if someone can point me in the right direction that would be fine.
thanks for all that you do.

Describe the problem

When I execute the script for v1.7.4, I get a 'NOT OK' message and none of the information will populate any of the fields. I only get 14 sensors discovered by telemetry and not the FM sensor. This 7” quad will get up to 19 GPS sats and fly’s fine.

I have another 7” quad running Betaflight 4.2.6 that connects to the same radio and crossfire transmitter. The script will run on It and picks up a lot of information probably cause it has 23 sensors discovered. 

Hardware and Software information

Inav 2.6.1
Holybro Kakute F7 v1.5 FC
TBS Crossfire diversity nano RX v4.11
Taranis X9D+ ACCST OpenTX 2.3.11
TBS Transmitter v4.11 
Beltian BN-880 GPS

hprompt, 2021-05-16:
how do I label this as a question and get it assigned

teckel12, 2021-05-25:
If you're not getting the `FM` sensors the problem is either with Crossfire or INAV settings.  `FM` is the flight mode, something is 'NOT OK' if there's no flight mode.

hprompt, 2021-05-25:
Hey Tim
thanks for assigning this and getting it started. I had also opened an issue inav:
TBS Crossfire Telemetry Sensors Problem in INav #6931.
I have passed my attached dump file around and just have run out of ideas.
if you can give me somethings to look at it would be great

thanks
Hayward

[dump.txt](https://github.com/teckel12/LuaTelemetry/files/6542807/dump.txt)

hprompt, 2021-06-08:
Hello Tim

1: Under "Output Map" for Diversity Nano RX set channels 3&4 to Mavlink TX & Mavlink RX.
2: Wired CH3 and CH4 to UART2.
3: On ports tab in inav, UART2 set it to Mavlink (telemetry column) - Made sure to use 57600 BAUD.
4: on Taranis X9D+ Delete all then Discover New Sensors.
I did this now I get a total of 22 Sensors:
1RSS,2RSS,RQly,RSNR,ANT,RFMD,TPWR,TRSS,TQly,TSNR
GPS,GSpd,ALT,SATS,PITCH,ROLL,YAW,Hdg,RxBT,CURR,CAPA,BAT_

I believe I'm missing the FM sensor, how would I get the FM sensor?

closed 2021-07-30 by stale[bot]

mactac, 2022-04-10:
same problem here, not stale.