teckel12/LuaTelemetry #463: INAV Lua text/graphics colour change on other widget screen

FLYN-001, 2021-05-14:
On the flexi or default theme, if text colour is anything other than black on a first widget screen with trims & sliders enabled, flicking between the inav Lua & back to the first widget screen changes text colour from white (on default) or user selected (on Flexi) to black, won't change back until reselect colour or new theme.

To Reproduce
Steps to reproduce the behavior:

select theme default or flexi.
change Text colour on flexi to anything other than black confirm & exit to 1st widget screen with sliders & trims on.
flick over screen to iNav Lua on second widget screen then back to 1st widget screen.
See error of graphics & text colour change to black even though theme set colour is anything but.

Expected behavior
colour of text & slider/trim graphics not to change colour

General information

OpenTX version 2.3.11
Hardware FRSKY X12S ACCESS
inav Lua 1.7.4

[image]
[image]
[image]

teckel12, 2021-05-25:
This is due to how the Horus works with widgets, they're not isolated, they can change the colors of other widgets.

But there's a solution, in the documentation:

https://github.com/teckel12/LuaTelemetry/wiki/Installation#installsetup-lua-telemetry-on-transmitter

> Optionally (while still selecting the iNAV script), long-press Enter, select Widget settings where you can enable Restore (to restore your theme's colors) and set your theme's Text color and Warning color

closed 2021-05-25 by teckel12

mactac, 2022-04-10:
The solution does not work, at least for me.