teckel12/LuaTelemetry #466: How to access settings menu on Jumper T-Lite?

avsaase, 2021-06-12:
### Describe your question
The title. As far as I can see there is no way to access the menu and change the settings on this transmitter.

### To reproduce
System and model buttons don't work. Neither do the stick commands used on the newer transmitters.

### Expected behavior
It should be possible to change the settings.

### Additional context
I would also be perfectly happy to change some text files to adjust the settings. `\SCRIPTS\TELEMETRY\iNav\cfg\readme.txt` suggests the configuration files for each model are stored in that folder, but the folder is otherwise empty.

Sekilsgs2, 2021-07-05:
This need some modifications to iNav.lua from src folder - 
in 511 line change line
`elseif event == MENU then`
to 
`elseif event == MENU or event == EVT_DOWN_LONG then`
Than need recompile this edited script in opentx simualtor and next place compilled script to Radio.

avsaase, 2021-08-19:
Still needs to be implemented.

julianschweizer, 2021-10-03:
I added a PR https://github.com/teckel12/LuaTelemetry/pull/476

avsaase, 2022-01-01:
not stale

axelfenner, 2022-01-19:
solve ths issue....

closed 2022-03-12 by stale[bot]