teckel12/LuaTelemetry #478: Widget issues with EdgeTX 2.5

Xedos9er, 2021-11-01:
### Is your feature request related to a problem? Please describe
After updating EdgeTX from 2.4 to 2.5 the INAV Widget on my TX16S doesn´t work properly. 
Not all informations are visible and background looks weird.

### Describe the solution you'd like
Make INAV Telemetry compatible with edgetx 2.5

### Additional context
The new edgetx update 2.5 makes some changes to widget touch compatibility. This seems to cause the issues.

FranzPVal, 2021-11-11:
I can confirm the issue:
All active sensor text seems to be invisible (or rendered with the same blue color as the background).
The artificial horizon is light blue (sky) and darker blue (ground) now.
At the configuration it get's worse: light blue text on light gray - hard to read even indoors with lights off. Selection is yello on light blue. Not optimal because the background is too bright.
---
Please fix. Maybe there is a change to work on the color scheme to bring more contrast to it...

dod22, 2021-11-15:
I can confirm this is issue as well.

closed 2022-01-06 by stale[bot]

mactac, 2022-04-10:
This is definitely still an issue, the lua script is essentially unusable on EdgeTX